moveit/moveit #3075: Impossible Motion Planned

DaveBGld, 2022-03-02:
### Description

It's a new Interbotix XV300S 6dof arm, new ROS Install. Running on the actual arm generates planning that is actually damaging to the arm.

After loading Moveit with the actual robot (se below for command), a small incremental movement will cause a movement plan with wild movements on the waist and forearm_roll motors to the point of disconnecting motor cables after so much rotation.

[Please watch this screen capture video that explains the problem](https://youtu.be/J9egEtAhqdA)

[And this is a second screen capture video of the same problem in a different pose](https://youtu.be/I6k3selIx5I)

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build,  release version 1.14.12 RViz version 1.13.23 (melodic).

### Steps to reproduce

Complete ROS install using instructions [at Interbotix](https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_interface/software_setup.html#amd64-architecture)

```sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh' > xsarm_amd64_install.sh
chmod +x xsarm_amd64_install.sh
./xsarm_amd64_install.sh
```
[Contents of xsarm_amd64_install.sh](https://github.com/ros-planning/moveit/files/8171872/xsarm_amd64_install.txt)

Connected the arm

executed:
`roslaunch interbotix_xsarm_moveit xsarm_moveit.launch robot_model:=vx300s use_actual:=true dof:=6
`
Select in RViz Query Goal State

Move the end effector incrementally small distances up the z axis and see the waist and forearm_roll motors move

### Expected behaviour

There should be minimal or no movement of the waist and forearm_roll motors

### Actual behaviour
The waist and forearm_roll motors rotate wildly

simonschmeisser, 2022-03-02:
This looks like a problem with either the inverse kinematics plugin or some self-collisions preventing the IK plugin from working well. Which one are you using? According to this https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/kinematics.yaml it's LMA. I have no experience with that one but can only recommend you to use a analytical solver like `opw` if the geometry allows it.

You could check what happens if you remove the "collision aware ik" checkbox and repeat your movement

DaveBGld, 2022-03-03:
I've tried LMA, KDL and TRACK_IK with  similar behavior

v4hn, 2022-03-04:
> It's a new Interbotix XV300S 6dof arm, new ROS Install. Running on the actual arm generates planning that is actually damaging to the arm.

That statement is absolutely wrong w.r.t. your video. You generate *IK* results, not a plan and definitely no movement! If you want to plan to go to the orange robot state, *click "plan"* in the rviz gui.
What you show in the video is simply that the configured IK solver for a single pose target might yield a solution in joint space that is far away from your current state.

DaveBGld, 2022-03-04:
Hi Michael, Thanks for your reply. The thing is that before that video, I did did a Plan and Execute to move the end effector to the same position as in the first video, and  and because of the spin in the forearm_roll and wrist_rotate a cable got disconnected, and the arm got disabled.

I understand the problem you have with my choice of words, in my attempt to explain my problem, but under that is the very real issue that the arm gets disabled by executing this very simple movement.

Very thankful for any help

DaveBGld, 2022-03-05:
Dear all, thank you for your support. I'm scheduled to travel for work the next two weeks, and I'll pick up this issue again when I return and report on the solution....