moveit/moveit #3077: Robot arm position calculation

simmnn, 2022-03-04:
### Description

We use an UR16e robot arm and want to move with our tool from left to right while holding the tool in height on zero. This is very import because we are removing material and we want to create a flat surface.

We have an imu on the robot that measures the angle of the robot and we have a laser receiver that measures how high above the surface we are. Both these components are mounted on our derivable robot, together with the robot arm.

the tf tree of the robot looks like this (simplified)
[image]

I recorded some cycles and i see strange things when i compare some signals:
The first signal is coming from the MoveGroupInterface where i use the function getCurrentPose() to get the current pose. 
The second signal is calculated using the tf tree: robot_base -> ground -> laser -> base_link -> base -> tool0_controller

Both signals should represent the same transform from robot_base to tool0. But when i plot them in time i see a difference between the two. (blue and red)
[image]

The x component seems to have a constant offset of +-250mm, the y component has an offset of 2mm, the z component seems to not have a constant offset: in the middle of the traject it comes closer to the other curve. This behavior repeats itself for all cycles.

Does anyone have an idea of where it all goes wrong? I have rosbags so i can provide some more signals if necessary.

Thanks in advance!

### Your environment
ROS Melodic
Moveit release tag 1.1.8

rhaschke, 2022-03-04:
Probably, MoveGroupInterface's `getCurrentPose()` returns the pose of a different link. Do you pass the correct link name?

simmnn, 2022-03-04:
> Probably, MoveGroupInterface's `getCurrentPose()` returns the pose of a different link. Do you pass the correct link name?

I use `move_group->setPoseReferenceFrame("robot_base");` So i think this is ok?

rhaschke, 2022-03-04:
> I use `move_group->setPoseReferenceFrame("robot_base");` So i think this is ok?

This just specifies the reference frame used to specify Cartesian targets.
I was talking about the `link name` argument to `getCurrentPose()`.

simmnn, 2022-03-04:
> > I use `move_group->setPoseReferenceFrame("robot_base");` So i think this is ok?
> 
> This just specifies the reference frame used to specify Cartesian targets. I was talking about the `link name` argument to `getCurrentPose()`.

I don't fill in the argument so i assume it gives me the transform to the link that it is planning to? In this case it is the tool0 link

rhaschke, 2022-03-04:
> I don't fill in the argument so i assume it gives me the transform to the link that it is planning to?

Don't speculate, read the docs:
```cpp
geometry_msgs::PoseStamped getCurrentPose(const std::string& end_effector_link = "") const;
```
> Get the pose for the end-effector `end_effector_link`.
> If `end_effector_link` is empty (the default value) then the end-effector reported by `getEndEffectorLink()` is assumed

```cpp
const std::string& getEndEffectorLink() const;
```
> Get the current end-effector link.
   This returns the value set by `setEndEffectorLink()` (or indirectly by setEndEffector()).
   If `setEndEffectorLink()` was not called, this function reports the link name that serves as parent
   of an end-effector attached to this group. If there are multiple end-effectors, one of them is returned.
   If no such link is known, the empty string is returned.

simmnn, 2022-03-07:
Thank you for the reply's. I'm closing this because after setting the correct links i still have the wrong tf but solved the problem in a different way

closed 2022-03-07 by simmnn