moveit/moveit #3080: CollisionObject pose is wrong (twice as far as it's supposed to be in rViz)

julien-audet, 2022-03-05:
### Description

When adding new collision objects (moveit_msgs::CollisionObject) using the planning scene interface, the position of the collision object is double (x = 2x, y = 2y, z = 2y) what it's supposed to be in rViz. It was fine before I decided to switch to Noetic from Kinetic. 

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Binary build (latest)
* 1.1.8-1focal.20220226.080138

### Steps to reproduce
```
// Collisions objects and colors vector
std::vector<moveit_msgs::CollisionObject> collision_objects;
std::vector<moveit_msgs::ObjectColor> object_colors;

moveit_msgs::CollisionObject object;
object.header.frame_id = "world";
object.id = "workpiece";

// Create mesh using any stl file (it also doesn't work with a primitive)
shapes::Mesh* mesh = shapes::createMeshFromResource
        ("package://interactive_finishing/meshes/generated_mesh.stl");
shapes::ShapeMsg mesh_msg;
shapes::constructMsgFromShape(mesh, mesh_msg);
shape_msgs::Mesh mesh_shape_msg = boost::get<shape_msgs::Mesh>(mesh_msg);

// Object pose
geometry_msgs::Pose object_pose;

// Rotation
Eigen::Quaterniond q{0.7071, 0, 0, 0.7071};
q.normalize();
object_pose.orientation.w = q.w();
object_pose.orientation.x = q.x();
object_pose.orientation.y = q.y();
object_pose.orientation.z = q.z();

// Translation
object_pose.position.x = 0.5;
object_pose.position.y = 0.5;
object_pose.position.z = 1;

object.meshes.push_back(mesh_shape_msg);
object.mesh_poses.push_back(object_pose);
object.pose = object_pose;
object.operation = object.ADD;

collision_objects.push_back(object);

moveit_msgs::ObjectColor object_color;
object_color.id = "workpiece";
object_color.color.a = 1;
object_color.color.r = 1;
object_color.color.g = 1;
object_color.color.b = 1;

object_colors.push_back(object_color);

// Apply collision objects to ROS planning scene
planning_scene_.applyCollisionObjects(collision_objects, object_colors);
```

### Expected behaviour and Actual behaviour
The object position is supposed to be (x,y,z) = (0.5, 0.5, 1) m, but it's actually (1, 1, 2) m when looking in rViz.
The object rotation is also wrong (twice the rotation). When I look at the topic /move_group/monitored_planning_scene, the collision object pose is right (0.5, 0.5, 1), so I have no idea why rViz is showing the wrong pose for the collision object. Thank you for your help.

simonschmeisser, 2022-03-05:
Please test with a sphere or box as well, I suspect your mesh is not centered actually

julien-audet, 2022-03-05:
Hi Simon, thanks for the quick reply. I actually tested it with a primitive (box), and the result was the same. But I just saw that I've used the pose twice in the definition of the collision object (in object.mesh_poses and object.pose), which is why I had that problem.
```
object.mesh_poses.push_back(object_pose);
// object.pose = object_pose; // I removed this line
```
Now everything is working! I'm closing this issue

closed 2022-03-05 by julien-audet