moveit/moveit #3094: Does `pilz_industrial_motion_planner` support `max_cartesian_speed`?

rr-tom-noble, 2022-03-16:
Hi,

I'm attempting to plan motions through pilz (PTP & LIN) using a desired tool speed rather than joint scaling factors. I've tried setting the `max_cartesian_speed` and `cartesian_speed_end_effector` links within the `MotionPlanRequest` msg (multiple since I'm trying to plan a sequence), however, I was getting the following error:

```
[ERROR] [1647425872.382288384] [/fake/move_group]: Velocity scaling not in range [0.0001, 1], actual value is: 0
[ERROR] [1647425872.617527319] [/fake/move_group]: Planner threw an exception (error code: -2): Could not solve request
```

I've tried setting the scaling factor to 1.0 which allows pilz to plan, but this causes the robot to always plan at full speed, seemingly ignoring the cartesian speed constraint. Does pilz support this option, or do all speeds need to be set via the scaling factors?

Thanks

simonschmeisser, 2022-03-16:
try `LANG=c` and see if that fixes the output

gautz, 2022-03-21:
You mean you are trying to use #2856 with Pilz? I only used this with OMPL. Pilz currently requires that the `max_velocity_scaling_factor `is set. So you should e.g. set ```
cartesian_limits:
  max_trans_vel: 1
```
and then compute the `max_velocity_scaling_factor =desired_cartesian_velocity / max_trans_vel`.

I don't plan to work on #2856 further, so maybe you could extend it to enable more intuitive Cartesian Speed setting with Pilz, and maybe #2935 and #2934.