moveit/moveit #3097: orientation constraint gets ignored

cs-99, 2022-03-18:
Hello,
since setting orientation constraints did not work a couple of times for me already, i made a minimal example for it.
After executing it a couple of times it looks like the constraint is simply being ignored sometimes (using rviz and gazebo).
Can somebody tell me the issue here? I am using ROS Noetic and Ubuntu 20.04.

My main:

	ros::init(argc, argv, "min_example_constraint");
	ros::NodeHandle node_handle;

	ros::AsyncSpinner spinner(1);
	spinner.start();
	ros::Duration(5).sleep();

    moveit::planning_interface::MoveGroupInterface move_group_interface("panda_arm");
    moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

	std::vector<double> joint_vals = move_group_interface.getCurrentJointValues();
	joint_vals[5] += 0.4;
	move_group_interface.setJointValueTarget(joint_vals);

	moveit::planning_interface::MoveGroupInterface::Plan rot_plan;
	move_group_interface.plan(rot_plan);
	move_group_interface.execute(rot_plan);
	
	geometry_msgs::PoseStamped pose_st = move_group_interface.getCurrentPose();
	pose_st.pose.position.x = -0.6;
	pose_st.pose.position.y = 0.0;
	pose_st.pose.position.z = 0.6;
	move_group_interface.setPoseTarget(pose_st);

	moveit_msgs::OrientationConstraint orientation_constraint;
	orientation_constraint.weight = 1.0;
	orientation_constraint.link_name = "panda_link8";
	orientation_constraint.header.frame_id = "world";
	orientation_constraint.orientation = pose_st.pose.orientation;
	orientation_constraint.absolute_x_axis_tolerance = 0.1;
	orientation_constraint.absolute_y_axis_tolerance = 0.1;
	orientation_constraint.absolute_z_axis_tolerance = 0.1;
	moveit_msgs::Constraints constraints;
	constraints.orientation_constraints.push_back(orientation_constraint);
	move_group_interface.setPathConstraints(constraints);

	moveit::planning_interface::MoveGroupInterface::Plan plan;
	move_group_interface.plan(plan);
	move_group_interface.execute(plan);

	ros::waitForShutdown();
	return 0;

My launch file is just spawning this node after an include of "$(find moveit_resources_panda_moveit_config)/launch/demo.launch".

v4hn, 2022-03-25:
For a minimal example please always provide a git repo with a full package and not a partial crop from a c++ file.
I tried to get this reproduced and am not 100% sure what you observe.

Most likely, this is the "usual error" about the planner planning in Cartesian space with projection to IK solutions (that might be far away...). It's an awful bug that these "solutions" are allegedly valid and it should have been fixed years ago. Feel free to dig into the OMPL planner interface to provide a patch.

If your constraints do not change dynamically and your state space is not extensive, the quickest solution is still [the constraint approximation db](https://ros-planning.github.io/moveit_tutorials/doc/planning_with_approximated_constraint_manifolds/planning_with_approximated_constraint_manifolds_tutorial.html) that samples the manifold in joint space beforehand.

If you are willing to invest more work, #2273 is a big effort to bring proper planning with path constraints into MoveIt, but it's not merged mostly for a shortage of reviewers.

closed 2022-03-25 by v4hn

cs-99, 2022-04-03:
@v4hn I have pushed the [repository](https://github.com/cs-99/min_example_constraint) to github now. What other options do I have to solve the problem except the [database](https://ros-planning.github.io/moveit_tutorials/doc/planning_with_approximated_constraint_manifolds/planning_with_approximated_constraint_manifolds_tutorial.html)?

yamachaso, 2022-11-26:
@cs-99 
Did you find any solution? I have encounterd the same problem.

cs-99, 2022-11-26:
@yamachaso Not really, we switched to ROS2 in the mean time.