moveit/moveit #3098: Can anyone provide any guidance on null space self motion?

Yitian-Xian, 2022-03-22:
Hi everyone, may I know if there is any method to program a self motion with the end-effector still using its null space? Thank you!

rhaschke, 2022-03-22:
I'm not sure what you mean by _self motion_. In any case, as far as I know, there is no API available to perform a nullspace-only motion. You could have a look at the source code implementing the joints tab in rviz:
https://github.com/ros-planning/moveit/blob/b6895ab72830d66c6382148f15674e7ab4f2cbac/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_joints_widget.cpp#L363-L382

The nullspace basis is computed here:
https://github.com/ros-planning/moveit/blob/b6895ab72830d66c6382148f15674e7ab4f2cbac/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_joints_widget.cpp#L301

v4hn, 2022-03-23:
I prepared a similar answer, but could not post it due to the github outage yesterday. I'll close this already because it's no direct issue in MoveIt.

closed 2022-03-23 by v4hn

v4hn, 2022-10-11:
Did you notice the "Nullspace exploration" sliders at the bottom of the joints tab in the RViz display?

[image]

[The implementation](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_joints_w
idget.cpp#L305-L338) is based on the Eigenvectors of the Jacobian computed by SVD.

Of course you would still need a time parameterization if you select waypoints like this.

Also you could solve IK for waypoints with additional constraints to move joints while retaining the eef pose. But this is no direct MoveIt question then.
Either way, this is not an actual issue with the project, so I'll close it.