moveit/moveit #3102: chomp_motion_planner build error

MHS-Gautam, 2022-03-25:
### Description

while building moveit from source I get the following error:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:187:48: error: expected ‘;’ at end of member declaration
   EIGEN_DEFAULT_COPY_CONSTRUCTOR(QuaternionBase)
                                                ^
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:188:64: error: expected ‘;’ at end of member declaration
   EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(QuaternionBase)

It seems like a compiler issue at first glance or something to do with C++14 but i'm not sure and cannot figure it out. This shows up while building the chomp_motion_planner

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build
* master branch
* C++Standard Used to build: C++14 (tried with C++17 but get the same error)

rhaschke, 2022-03-25:
Looks like you screwed your system. Our [CI build succeeds without problems](https://github.com/ros-planning/moveit/actions/runs/2028374768). Hence, I'm afraid we cannot really help.
Do you use any macro definitions screwing up the existing code? Did you try cleaning your workspace? Try a clean build!

MHS-Gautam, 2022-03-25:
I see that the issue arises on the robot_model.h reference which is part of moveit_core but moveit_core builds successfully and it fails to build the chomp motion planner so it's really weird. I tried clean build but to no avail. I also tried to test some sample quaternion code and i was able to run that as well in the same environment.

Pinklet, 2024-03-28:
> [/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:188:64: error: expected ‘;’ at end of member declaration EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(QuaternionBase)] I have the same problem and would like to seek help! Have you solved this problem?

你好 我也遇到了这个问题！请问你当时解决了吗

> [/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:188:64: error: expected ‘;’ at end of member declaration EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(QuaternionBase)] I have the same problem and would like to seek help! Have you solved this problem?

你好 我也遇到了这个问题！请问你当时解决了吗