moveit/moveit #3110: No controller and joint detected

arnauddomes, 2022-04-05:
### Description
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* LibFranka 
* Ros_Franka
* MoveIt

I try to run MoveIt on my Franka Emika robot. I run that command to run a controller and joint publisher

`roslaunch franka_control franka_control.launch robot_ip:=172.16.0.2 
`

Then I run MoveIt with that command and ros_control

`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true
` 

Here is the published topic:
```
/attached_collision_object
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/franka_control/error_recovery/cancel
/franka_control/error_recovery/feedback
/franka_control/error_recovery/goal
/franka_control/error_recovery/result
/franka_control/error_recovery/status
/franka_gripper/grasp/cancel
/franka_gripper/grasp/feedback
/franka_gripper/grasp/goal
/franka_gripper/grasp/result
/franka_gripper/grasp/status
/franka_gripper/gripper_action/cancel
/franka_gripper/gripper_action/feedback
/franka_gripper/gripper_action/goal
/franka_gripper/gripper_action/result
/franka_gripper/gripper_action/status
/franka_gripper/homing/cancel
/franka_gripper/homing/feedback
/franka_gripper/homing/goal
/franka_gripper/homing/result
/franka_gripper/homing/status
/franka_gripper/joint_states
/franka_gripper/move/cancel
/franka_gripper/move/feedback
/franka_gripper/move/goal
/franka_gripper/move/result
/franka_gripper/move/status
/franka_gripper/stop/cancel
/franka_gripper/stop/feedback
/franka_gripper/stop/goal
/franka_gripper/stop/result
/franka_gripper/stop/status
/franka_state_controller/F_ext
/franka_state_controller/franka_states
/franka_state_controller/joint_states
/franka_state_controller/joint_states_desired
/joint_states
/move_group/cancel
/move_group/display_contacts
/move_group/display_cost_sources
/move_group/display_grasp_markers
/move_group/display_planned_path
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/motion_plan_request
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_pipelines/ompl/ompl/parameter_descriptions
/move_group/planning_pipelines/ompl/ompl/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/rosout
/rosout_agg
/rviz_mdt_6591_4173660562006021822/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_mdt_6591_4173660562006021822/motionplanning_planning_scene_monitor/parameter_updates
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/rviz_visual_tools
/rviz_visual_tools_gui
/sequence_move_group/cancel
/sequence_move_group/feedback
/sequence_move_group/goal
/sequence_move_group/result
/sequence_move_group/status
/tf
/tf_static
/trajectory_execution_event
```

Here is the full log for MoveIt command:
https://www.codepile.net/pile/OVqRgz93

And the trouble at the end of file:

```
[ INFO] [1649067874.512278787]: SimpleSetup: Path simplification took 0.007065 seconds and changed from 3 to 2 states
[ERROR] [1649067874.514458067]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ERROR] [1649067874.514557474]: Known controllers and their joints:

[ERROR] [1649067874.514625791]: Apparently trajectory initialization failed
[ INFO] [1649067874.533892635]: ABORTED: Solution found but controller failed during execution
```

### Expected behaviour
I should see robot moving when I move it in Rviz.

### Actual behaviour
The robot position is well displayed but the robot is not moving.

arnauddomes, 2022-04-05:
I will try. I just try to find a simple example of moveit controller with panda franka emika arm.
It seems it comes more from the command I launch than the code.

Is there a clear working example or tutorial of Moveit with Panda arm?

rhaschke, 2022-04-05:
This is a config issue and should be reported at [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config).
Which branch of panda_moveit_config do you use? Try the noetic-devel branch, please.