moveit/moveit #3111: A delay between each execution in moveit

omar1slam, 2022-04-07:
I have a three dof robot, and i'm using setJointValueTarget to change joint values and plan and execute the motions, it works fine but there's a delay between each motion and other which make s the robot feel sluggish. I understand that the move function is a blocking function and it's probably the planning requests that take time. I tried the async move fucntion but it skips through a lot of motions which isn't ideal. So if anyone could help out on how to make my motions smoother and faster in execution.

cambel, 2022-04-08:
@omar1slam  You can see this [example](https://github.com/o2ac/o2ac-ur/blob/74c82a54a693bf6a3fc995ff63e7c91ac1fda6fd/catkin_ws/src/o2ac_routines/src/o2ac_routines/robot_base.py#L574) where we do exactly what @rhaschke said.
First we plan each trajectories alone then execute them together in series. Taking in to account that the 2nd point starts at the 1st plan last point and same fo ther 3rd point. However, that only moves the delay to the beginning.

For something fancier results, you can follow a more complicate approach of planning the first motion and execute it, then use the expected duration of the trajectory to simultaneously plan the 2nd point, then execute it after point 1st has been completed, and repeat for the 3rd one and so on.

A concrete implementation of that approach is [here](https://github.com/o2ac/o2ac-ur/blob/74c82a54a693bf6a3fc995ff63e7c91ac1fda6fd/catkin_ws/src/o2ac_routines/src/o2ac_routines/base.py#L1540)

omar1slam, 2022-04-18:
Thanks guys for your replies, I managed to make a series of points and it plans but I can't execute. I don't know one is supposed to go about executing a trajectory. If you could give some input I'd be grateful