moveit/moveit #3117: How do I control the speed of a real robot with MoveIt  ( in ComputecartesianPath) ?

AtariaLee, 2022-04-24:
I had control a real robot(fanuc m710ic_50) with Moveit. And i created a plan path using  `computeCartesianPath` , its worked in my robot. But the robot moves too slowly ,so i  tried to speed up my robot.  I changed some parameters using`arm.setMaxVelocityScalingFactor(0.6)` and ` arm.setMaxAccelerationScalingFactor(0.4)` ,but it didnt work. 
Then I created a function to try to redefine the parameters in my plan path, like
```
 void scale_trajectory_speed(moveit::planning_interface::MoveGroupInterface::Plan &plan, double scale)

    {
        int n_joints = plan.trajectory_.joint_trajectory.joint_names.size();
        
        for(int i=0; i<plan.trajectory_.joint_trajectory.points.size(); i++)
        {
            plan.trajectory_.joint_trajectory.points[i].time_from_start *= 1/scale;
            
            for(int j=0; j<n_joints; j++)
            {
                plan.trajectory_.joint_trajectory.points[i].velocities[j] *= scale;
                plan.trajectory_.joint_trajectory.points[i].accelerations[j] *= scale*scale;
            }
        }
    }
```
unfortunately, it didt work too .So how could i control the speed of realrobot with Moveit (using computeCartesianPath)?
Thanks for everyones help!

AtariaLee, 2022-04-24:
Then i tried to use `trajectory_processing::IterativeParabolicTimeParameterization::computeTimeStamps` to change time scale,like 

    moveit::planning_interface::MoveGroupInterface::Plan plan;

    robot_trajectory::RobotTrajectory rt(arm.getCurrentState()->getRobotModel(),"manipulator");

    rt.setRobotTrajectoryMsg(*arm.getCurrentState(),trajectory);
    trajectory_processing::IterativeParabolicTimeParameterization iptp;
    bool ItSuccess = iptp.computeTimeStamps(rt);
    ROS_INFO("Computed time stamp %s", ItSuccess ? "SUCCEDED" : "FAILED");
    rt.getRobotTrajectoryMsg(trajectory);
    plan.trajectory_=trajectory;

but it didnt work too. Am I using it the wrong way ?? Please help me ,I've been puzzling over this question for days. I am truly very grateful for your help. !!

simonschmeisser, 2022-04-24:
Typically you need to set acceleration limits in `joint_limits.yaml` to get realistic accelerations. You can compare the `time_from_start` of your trajectory (by using `rostopic echo /joint_trajectory_action/goal` (use tab completion to find the actual name of the topic) with the time it actually takes for the robot to move. If the robot takes a lot longer to move than calculated then you have some issue with the driver. If the times somewhat match but are too long you will need to increase the acceleration limit (the default is `1/s`)

AtariaLee, 2022-04-24:
but I didn't set the  limits in `joint_limits.yaml`,
```
joint_limits:
  joint_1:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
  joint_2:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
  joint_3:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
  joint_4:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
  joint_5:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
  joint_6:
    has_velocity_limits: false
    max_velocity: 0
    has_acceleration_limits: false
    max_acceleration: 0
```

simonschmeisser, 2022-04-24:
Not setting it means the default of `1/s` is used, which is slow. Try something in the range of 5 to 20. Unfortunately manufacturers do not provide the actual limits (and they are load and posture dependent) so you have to experiment a bit

AtariaLee, 2022-04-24:
thanks, i will try it tomorrow.
and Is that code correct ?
```

moveit::planning_interface::MoveGroupInterface::Plan plan;

robot_trajectory::RobotTrajectory rt(arm.getCurrentState()->getRobotModel(),"manipulator");

rt.setRobotTrajectoryMsg(*arm.getCurrentState(),trajectory);
trajectory_processing::IterativeParabolicTimeParameterization iptp;
bool ItSuccess = iptp.computeTimeStamps(rt);
ROS_INFO("Computed time stamp %s", ItSuccess ? "SUCCEDED" : "FAILED");
rt.getRobotTrajectoryMsg(trajectory);
plan.trajectory_=trajectory;

```

AtariaLee, 2022-04-25:
After i tried to change joint limits ,like
```
joint_limits:
  joint_1:
    has_velocity_limits: true
    max_velocity: 13.0543
    has_acceleration_limits: true
    max_acceleration: 3
  joint_2:
    has_velocity_limits: true
    max_velocity: 13.0543
    has_acceleration_limits: true
    max_acceleration: 3
  joint_3:
    has_velocity_limits: true
    max_velocity: 13.0543
    has_acceleration_limits: true
    max_acceleration: 3
  joint_4:
    has_velocity_limits: true
    max_velocity: 14.3633
    has_acceleration_limits: true
    max_acceleration: 3
  joint_5:
    has_velocity_limits: true
    max_velocity: 14.3633
    has_acceleration_limits: true
    max_acceleration: 3
  joint_6:
    has_velocity_limits: true
    max_velocity: 16.1959
    has_acceleration_limits: true
    max_acceleration: 3
```

Then came a warning    `Joint velocity-limits unspecified.  Using default velocity-ratio.`.
im confused that I had set the limit, why does the screen show an unspecified limit value?
So ,how could i set  `joint the velocity-limits`?

dthung99, 2024-05-25:
I had the same problem. I did solve it by modifying the **moveit_msgs::RobotTrajectory** msg. Specifically the **.joint_trajectory.points[i].time_from_start**. In case your robot have multi DOF joints, you can modify the **.multi_dof_joint_trajectory.points[i].time_from_start** instead

Here is my code:
```
void set_execute_duration_for_trajectory(moveit_msgs::RobotTrajectory& input_trajectory, double execute_duration) {
  std::vector<trajectory_msgs::JointTrajectoryPoint>& joint_trajectories = input_trajectory.joint_trajectory.points;
  size_t size = joint_trajectories.size();
  execute_duration = execute_duration/size;
  for (int i=0; i<size; i++) {
    joint_trajectories[i].time_from_start = ros::Duration(execute_duration*i);
  }

int main(int argc, char** argv)
{
// ...
// your main code

  moveit::planning_interface::MoveGroupInterface::Plan my_plan; 
  fraction = move_group.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory);
  set_execute_duration_for_trajectory(trajectory, execute_duration); // Slow down or speed up the robot
  my_plan.trajectory_ = trajectory

// your main code
// ...
}
```