moveit/moveit #3118: The issue of moveit_setup_assistant gui

darrenjiayinxie, 2022-04-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build - master branch 

### Steps to reproduce
Using the setup assistant for a robot arm, and when defining the robot pose, the joint label, joint value and slide bar become extremely small. 
 
### Expected behaviour
[image]

### Actual behaviour
The joint label, joint value and slide bar become extremely small. 
[image]

### Steps to fix the bug
Modify the source code '/moveit/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp':
modify line 525 and 831
![code_1](https://user-images.githubus
[image]
ercontent.com/16159311/165002505-1f34e4d0-dd92-41c4-9443-c0e2b4a5e36c.png)

rhaschke, 2022-04-25:
I cannot confirm this issue. Do you use a high-resolution screen?

darrenjiayinxie, 2022-04-25:
> That's what I expected. The actual problem is that the window sizes are absolute. They should be computed relative e.g. to the font size or to the main window.

I agree. I think this should be fixed.

PR #3219 (2022-09-16): MSA: Generalize sizing of joint list widgets
Replace fixed with adaptive sizing, based on available space and font metrics.
Fixes #3118.
closed 2022-09-16 by rhaschke