moveit/moveit #3121: ikfast  error : "symbol lookup error"  and  "undefined symbol"

lotus824, 2022-04-25:
Hello everyone!
I got issues when I use ikfast plugin on my program.

[ INFO] [1650895524.914301655]: Using empty link_prefix.
/opt/ros/melodic/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/lotus/catkin_ws/devel/lib//libprobot_anno_gripper_arm_moveit_ikfast_plugin.so: undefined symbol: _ZN23probot_anno_gripper_arm17GetFreeParametersEv
[move_group-8] process has died [pid 11970, exit code 127, cmd /opt/ros/melodic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/lotus/.ros/log/bf6f12d4-c4a0-11ec-9c12-9822eff17c1f/move_group-8.log].
log file: /home/lotus/.ros/log/bf6f12d4-c4a0-11ec-9c12-9822eff17c1f/move_group-8*.log
[INFO] [1650895524.994367, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1650895525.013073, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1650895525.063915374]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1650895525.063967391]: Using empty link_prefix.
/home/lotus/catkin_ws/devel/lib/probot_anno_gripper_grasping/grasping: symbol lookup error: /home/lotus/catkin_ws/devel/lib//libprobot_anno_gripper_arm_moveit_ikfast_plugin.so: undefined symbol: _ZN23probot_anno_gripper_arm17GetFreeParametersEv
[ INFO] [1650895525.065495915]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1650895525.150222628]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1650895525.151572157]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[grasping-11] process has died [pid 11975, exit code 127, cmd /home/lotus/catkin_ws/devel/lib/probot_anno_gripper_grasping/grasping __name:=grasping __log:=/home/lotus/.ros/log/bf6f12d4-c4a0-11ec-9c12-9822eff17c1f/grasping-11.log].
log file: /home/lotus/.ros/log/bf6f12d4-c4a0-11ec-9c12-9822eff17c1f/grasping-11*.log

JafarAbdi, 2022-04-25:
Hi @lotus824,

It can't find `probot_anno_gripper_arm::GetFreeParameters()`, can you provide more info? Is it possible to share your CMakeLists.txt file? or a reproducible example?

rhaschke, 2022-04-29:
Could you please provide more details on which type of IK fast solver you use?
I remember that not all advertised solver types are correctly implemented...

lotus824, 2022-04-29:
I seem to have solved the problem.  I tried to add this ```IKFAST_API int* GetFreeParameters() { return 0; }``` in ```probot_anno_gripper_arm_ikfast_solver.cpp```. IKFast plugin seems to work fine.