moveit/moveit #3122: How describe a custom tool URDF in Ros?

Muhrobot, 2022-04-26:
I already have an URDF model and it have 6 joints. My last link is flange, and now i want to build a eef_link to change the reference of the flange to my new tool. I try to create a tool insede my urdf file:

```
<joint name="flange_tool" type="fixed">
    <parent link="flange" />
    <child link = "tool" />
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0." />
    <axis xyz="1 0 0" />
</joint>

<link name="tool">
    <visual>
        <origin rpy="0.0 1.5708 0.0" xyz="0.064 0 0" />
        <geometry>
            <cylinder length="0.1" radius="0.01"/>
        </geometry>
        <material name="LightGrey">
            <color rgba="0.7 0.7 0.7 1.0" />
        </material>
    </visual>
    <collision>
        <origin rpy="0.0 1.5708 0.0" xyz="0.064 0 0" />
        <geometry>
            <cylinder length="0.1" radius="0.01"/>
        </geometry>
    </collision>
</link>
```
It show's on Rviz or Gazebo simulation, but the TCP coordenades do not change! So, how to do a generic TCP for my robot?

rhaschke, 2022-04-27:
You also need to adapt your robot´s SRDF to include your new link in the kinematic chain of your planning group.

Muhrobot, 2022-05-02:
> You also need to adapt your robot´s SRDF to include your new link in the kinematic chain of your planning group.
@rhaschke Thanks for the awsner! I update a SDRF and others .yaml archives and it work for me! Thanks :)

closed 2022-05-02 by Muhrobot