moveit/moveit #3127: PlanningSceneMonitorPtr initialize time

emielke12, 2022-05-03:
### Description

Recently updated some code running on kinetic to now use noetic. I have a line that did not change at all in the update seen here:
`planning_scene_monitor::PlanningSceneMonitorPtr planning_scene_monitor_ptr = std::make_shared<planning_scene_monitor::PlanningSceneMonitor>(robot_description_);`

This line, in kinetic, took <1 second to run, but is now taking >7 seconds. Not sure why this would be the case, or how it is happening.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Binary build 1.1.9-1focal.20220413.195714

rhaschke, 2022-05-03:
Usually, the most time-consuming part in PSM initialization is loading the robot model (and the associated IK plugins).
Did this change for you?

emielke12, 2022-05-03:
the robot model did not change (to my knowledge), the IK plugin being used is trac_ik, which updated from the kinetic to noetic versions.

rhaschke, 2022-05-03:
It might be helpful to enable debug logs and observe the time stamps printed for each debug message to further narrow down the slowdown...

emielke12, 2022-05-09:
I tracked down the major time-suck to [this line](https://github.com/ros-planning/moveit/blob/dd0fb2a3a91f2dad7275c70efd52754a01517116/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L198) in the `planning_scene_monitor`. 

It doesn't appear that `rm_loader_->getModel()` takes too much time, as it is called just a couple lines above there, and that doesn't take longer than a few ms.

emielke12, 2022-05-09:
So, from there it looks like its heavy into the FCL world. I also had to update other related code from fcl v0.5 to fcl v0.6, and have been having similar issues with slow downs there, so I'm not shocked to see that it is causing slow-downs in moveit as well. 

Maybe this is now something to bring up in fcl. I'll post there as well and see what happens.

@rhaschke  If you have any other ideas, let me know.

rhaschke, 2022-05-12:
Thanks for tracking this down to FCL and filing an issue there. If allocating an FCL environment takes much more time now, that‘s really bad news.

simonschmeisser, 2022-05-23:
During construction of the FCL Collision Detector it creates a set of collision objects for the links of your robot. This takes about 30ms for my test scene but strongly depends on the size of your meshes. I don't see anything taking unusually long here (I was experimenting with orbit by google here, a profiler that I do not yet fully understand ...)