moveit/moveit #3138: IK with Robotiq gripper 2f 85 attached on UR5e

rrapi, 2022-05-17:
Hello everyone,

I am a master science student and for my thesis I am working on a UR5e robot with a Robotiq 2f 85 gripper directly attached.
In this project I have to grasp some objects in different positions and orientations, so for this reason I am using MoveIt to get directly the Inverse Kinematics of a  desired position and orientation depending on the object detected.
But the problem is that MoveIt considers as the reference frame of the endeffector the original one that ends in tool0, but as I am working with the gripper, I need that the referenze frame of the ee has to be of the robotiq gripper, so tranlates of ~19 cm wrt z-azis of previous one.
Without this setting the IK is always wrong as it changes completely the result.
Unfortunately ur5e_moveit_config package of has no gripper in the configuration, i tried to make one by me but did not work.

Could you please help me to solve this problem?
Thank you very much in advance!

Best regards,
Franci

rhaschke, 2022-05-18:
Please follow the SRDF tutorial to define your planning group. The arm group needs to include the end-effector frame.
If you don't succeed, explicitly state your problem. "It did not work" doesn't help to pinpoint your issue.
In general, this question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here.

rrapi, 2022-05-20:
Hi @rhaschke! Thank you for your answer! I solved that issue modifying the ur_macro.xacro file.

But now I have another issue when using moveit to send the gripper in a desired Pose and I get the following error: ABORTED: No motion plan found. No execution attempted. I tried various planners as RRTConnect, PRM or TRRT but no one worked.
I tried also with cartesian path but the final position of the robot is wrong.

Could you help me please? I don't know very well this topics and so I don't have any idea of what the problem could be.

rhaschke, 2022-05-20:
Probably the orientation of your target pose strongly differs from your ee pose, enforcing rather large joint motions to match them

rrapi, 2022-05-20:
And which is the reason that this happened? And how I can solve/avoid this??

rhaschke, 2022-05-21:
It is your job to specify the correct target orientation as well - not only target position.