moveit/moveit #3139: Link not found in model -- attached collision object computeCartesianPath

emielke12, 2022-05-17:
### Description

I'm following [this tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_interface/move_group_interface_tutorial.html) to attempt to pick up an object and [this tutorial](https://ros-planning.github.io/moveit_tutorials/doc/subframes/subframes_tutorial.html?highlight=subframe) to plan using the subframe as the frame of interest. 

Everything works fine if I follow verbatim, and using the`plan` and `execute` interfaces of move group. However, if I try to run `computeCartesianPath` I get the following error:
`LInk 'mylink/mytcp' not found in model 'fanuc_crx10ial'`. 

I'm trying to get the robot to move to a series of poses (doesn't necessarily need to be a cartesian path, but it can't be PTP that I have to run `plan` and `execute` over and over). If there is a way to do this without needing to call `computeCartesianPath` that would be acceptable as well, but I am wondering why the link is not added to the robot model after following the steps as specified in the tutorial?

### Your environment
* ROS Distro: [Noetic
* OS Version: Ubuntu 20.04
* Latest binaries

### Steps to reproduce
Follow subframes tutorial from moveit, but instead of using `plan` to plan path, do `computeCartesianPath`.

### Expected behaviour
I should get the planned cartesian path

### Actual behaviour
It should fail with an error `Link 'link' not found in model 'robot_model'`

rhaschke, 2022-05-18:
To simplify reproducing your issue, could you please create a new test in [subframes_test.cpp](https://github.com/ros-planning/moveit/blob/03bd588bcec36fc012bf5de56fa6c362288a579e/moveit_ros/planning_interface/test/subframes_test.cpp) using `computeCartesianPath()` and the Panda robot from `moveit_resources_panda_moveit_config`? Thanks.

emielke12, 2022-05-18:
@rhaschke Ok, done. I'm not sure the best method for getting this in for reproducing. What do you recommend?

rhaschke, 2022-05-18:
Please just file a PR with your unit test illustrating the issue and refer to this issue.

emielke12, 2022-05-18:
@rhaschke  #3143