moveit/moveit #3140: Constraint databases in MoveitCpp

HumbertoE, 2022-05-17:
## Description

Hello. I have a doubt I would like to ask for help with.

Following the [Planning with approximated constraint manifolds tutorial](https://ros-planning.github.io/moveit_tutorials/doc/planning_with_approximated_constraint_manifolds/planning_with_approximated_constraint_manifolds_tutorial.html) I was able to create the constraint database I needed. For planning and to be able to access more functionalities, I am using MoveitCpp directly instead of the Move Group C++ Interface.

How could I load and use the database with MoveitCpp? In #2941 [v4hn](https://github.com/v4hn) mentions the move_group node loads the database on startup, but I can't find how is this done and how to apply it for MoveitCpp.

Thank you in advance for the help.

simonschmeisser, 2022-05-22:
Hi,
your best bet is to search the code for MoveGroup for the section that loads the database and then see if you can find similar code in MoveItCpp. If not, please open a PR (no worries about quality etc. copy paste is fine, we can then discuss the details in the PR). You can also add a link here to where it's loaded in MoveGroup if you get stuck on the PR, that would save us some time already

HumbertoE, 2022-05-23:
Thank you for the answer. I will continue looking in the MoveGroup code and try to find it in the MoveItCpp code too and, as recommended, open a PR if I can't find it. I will write updates on this issue as I progress.

HumbertoE, 2022-05-31:
I think the database loading should be called after calling the planning plugin and therefore should be available, but I don't get it to work.

The constraints are loaded in [ompl_interface::ModelBasedPlanningContext::configure](https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L107) that is used in the following method/class that is used in the next one and so on:

- [ompl_interface::PlanningContextManager::getPlanningContext](https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/planning_context_manager.h#L181) (according to its description) 
- [ompl_interface::OMPLInterface::OMPLInterface](https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_planners/ompl/ompl_interface/src/ompl_interface.cpp#L50)
- [OMPLPlannerManager::initialize](https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_planners/ompl/ompl_interface/src/ompl_planner_manager.cpp#L107)

The last one overrides [PlannerManager::initialize](https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_core/planning_interface/include/moveit/planning_interface/planning_interface.h#L166) that, according to its description, is "called after the construction of the plugin, before any other call is made".

I might be using some namespace wrong, but the tutorial says to load the constraint database path as a parameter "move_group/constraint_approximations_path" so since I create the MoveItCpp object in a node called "trajectory_planning", "trajectory_planning/constraint_approximations_path" should work I think.

Is there something I am missing? If it is not a namespace error or a configuration error from my side, I can create the Pull Request.

HumbertoE, 2022-06-01:
I was analyzing the code more closely (still with MoveGroup instead of MoveitCpp) and I found the [OMPLInterface initializer](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_planners/ompl/ompl_interface/src/ompl_interface.cpp#L50) is called using ROS parameters, but the [getPlanningContext](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_planners/ompl/ompl_interface/src/ompl_interface.cpp#L108) method is never called.

I also added some log prints to make sure of this and also added some to [ConstraintsLibrary::loadConstraintApproximations](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_planners/ompl/ompl_interface/src/detail/constraints_library.cpp) and [ModelBasedPlanningContext::loadConstraintApproximations](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L875) where the constraint library should be loaded  and find out they don't execute at all, not even to try to load the ROS parameter storing the path to the constraints library.

Is the feature to use constraint libraries still active?

HumbertoE, 2022-06-06:
I found the correct namespace by setting the node logger level to debug. There is a log message that says the parameter was not found.

Using the planner in the move_group node, the parameter was: /move_group/planning_pipelines/ompl/constraint_approximations_path

and for my case, it was: /trajectory_planning/ompl/constraint_approximations_path

I will close the issue since it was just an incorrect namespace from my side

closed 2022-06-06 by HumbertoE