moveit/moveit #3144: Executing a trajectory with several pose targets

omar1slam, 2022-05-19:
Hi there, So I followed the moveit tutorial Motion planning pipeline on my own robot and the all my goal poses plan in sequence on rviz using `display_publisher.publish(display_trajectory);` just fine, now I'm trying to execute this plan on my actual hardware the ame as you would with` move()` or `execute()`. I tried several ways but can't seem to get it to work. Also I have an issue where all my points in the generated trajectory have `time_from_start = 0` and when I try to add the time manually it still doesn't work. If anyone could shed some light on how to properly execute a trajcetory I would be grateful.

JafarAbdi, 2022-05-19:
> I tried several ways but can't seem to get it to work

It would be great if you could share the logs or the error you're getting

> Also I have an issue where all my points in the generated trajectory have time_from_start = 0

This's probably due to you missing parameterization algorithm adapter (e.g. https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/ompl_planning_pipeline.launch.xml#L9) when starting your move_group node, could you please share your robot's moveit config package?

omar1slam, 2022-05-19:
Hi Jafar thanks for your quick reply,

> > I tried several ways but can't seem to get it to work
> 
> It would be great if you could share the logs or the error you're getting
I'm not getting an error actually it says execution completed but the robot doesn't move,
That's why I want to know how you're supposed to execute a trajectory because I'm probably doing it wrong

> This's probably due to you missing parameterization algorithm adapter (e.g. https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/ompl_planning_pipeline.launch.xml#L9) when [...]

Here you can find in this link my corresponding launch file to the one you're asking about [https://drive.google.com/file/d/1zJm2oHHHIl3WzR-FBigoCyxPklrKSZze/view?usp=sharing](url)

anubhav-dogra, 2022-05-26:
Hi, I am actually trying the same here. I'm not finding the way to execute those planned trajectories? can you share resources where I can learn about that ? 
which particular command will go for it ? 
Thanks