moveit/moveit #3153: # Issues with Moveit Servoing when running with Universal Robots ROS Driver on E series

Mohatashem, 2022-06-01:
Hello,

I've been trying to implement moveit_servo with the Universal Robots ROS driver for a UR10e robot. I was able to run the `/servo_server` node and publish to `servo_server/delta_twist_cmds` topic as well as run the `pose_tracking_example` file successfully with a gazebo simulation. 

However, I run into issues with implementation of `pose_tracking_example` on the real robot. 
## Steps followed:
1. Launching the bringup file
2. Launching the moveit_planning_execution launch file
3. Swithcing from trajectory controller to joint group position controller
4. Launching the pose_tracking launch file

## Behaviour:
- Pose tracking node seems to publish to `servo_server/target_pose`. In the video (visualization of the real robot), the moving frame is the target_pose. 

https://user-images.githubusercontent.com/40743268/171396092-f7dd77c7-2a39-4f07-8d6e-27cded110162.mp4

The code is the same as pose_tracking_example.cpp. the only changes I made are that there is an offset in z direction and then the target pose is decremented in the z-direction as well.

- However, the robot does not move at all. The terminal displays a `target control rate missed` and exits the program.

`rostopic echo /servo_server/delta_twist_cmds` shows zero values on all `/servo_server/delta_twist_cmds/twist/linear/...` and `.../angular`

## Details:
Here is the rqt_graph and as far as I've understood, everything seems to be connected:

[image]

Here is what I have so far:
1. The robot config yaml file:

``` 
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment
command_in_type: "speed_units" # "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/
scale:
    rotational:  0.3 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
    joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

publish_period: 0.002 # 1/Nominal publish rate [seconds]
low_latency_mode:  false #false #false # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

command_out_type: std_msgs/Float64MultiArray 

publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

move_group_name:  wall_arm 
planning_frame: wall_arm_base_link 

ee_frame_name: wall_arm_tool0   
robot_link_command_frame:  wall_arm_tool0 

incoming_command_timeout:  0.1
num_outgoing_halt_msgs_to_publish: 4

lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: joint_states
status_topic: status # Publish status to this topic
command_out_topic: /wall_arm/joint_group_pos_controller/command 

check_collisions: false # true # Check collisions?
collision_check_rate: 40 # [Hz] Collision-checking can easily bog down a CPU if done too often.

collision_check_type:  threshold_distance #threshold_distance
[... 100 lines cut ...]
[ INFO] [1653917663.782410623] [/servo_server] [ros.moveit_core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1653917667.711158390] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Starting planning scene monitor
[ INFO] [1653917667.712774088] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Listening to '/planning_scene'
[ INFO] [1653917667.712793934] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1653917667.714130366] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Listening to '/collision_object'
[ INFO] [1653917667.715539130] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1653917667.718427493] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[ WARN] [1653917668.044454951] [/servo_server] [ros.moveit_servo.servo_calcs]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1653917668.049594692] [/servo_server] [ros.moveit_servo.pose_tracking]: Target control rate was missed
[ INFO] [1653917668.052517787] [/servo_server] [ros.zerog_moveit_servo.pose_tracking_example]: Servo status: No warnings
[ INFO] [1653917678.064269919] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Stopping world geometry monitor
[ INFO] [1653917678.064670070] [/servo_server] [ros.moveit_ros_planning.planning_scene_monitor]: Stopping planning scene monitor
[servo_server-1] process has finished cleanly
log file: /home/pp1744/.ros/log/feae6ca2-e01a-11ec-a6d0-a0291986156e/servo_server-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I am not sure what to do next. Just for additional information, I tried running the servo_server node directly and published to `servo_server/delta_twist_cmds` directly via `rqt_ez_publisher`. In this situation, the robot seems to move albeit in a slow, jerky manner.

Any suggestions how to proceed from here?

Any help is really appreciated. I've been stuck at it for quite a while.

Mohatashem, 2022-06-01:
I confirm that I am using `threshold_distance`.

Stephen-Tellis, 2022-06-03:
Hi @Mohatashem,
I see that you have set both the `ee_frame_name` and the `robot_link_command_frame` parameters to be  _wall_arm_tool0_ .
Have you tried setting the `robot_link_command_frame` same as the planning frame?

VJan2022, 2024-09-27:
Hi, Did you solve this problem? I can use pose_tracking_exmple.launch to control the robot. However, the moving pose is not totally following my goal pose.