moveit/moveit #3155: [moveit_servo] Singularity Avoidance for MoveIt Servo

playertr, 2022-06-04:
**Is your feature request related to a problem? Please describe.**
When performing autonomous visual servoing with moveit_servo, I find that my robot arm often gets stuck in joint singularities. This is similar to a discussion on Damped Least Squares on https://github.com/ros-planning/moveit/issues/2617 .

**Describe the solution you'd like**
Implementing Damped Least Squares [1, eq. 10.18] could be a good way to avoid joint singularities. DLS adds a regularizing term to the Jacobian to avoid high joint velocities, which ends up avoiding singularities. However, DLS has several parameters and can result in unexpected arm motions, so incorporating DLS should be done carefully if at all.

I think that a plugin-based system of servoing backends could be an elegant solution, to allow people to pick the servoing backend they desire. This would also allow servo backends with other nifty behavior, like avoiding joint limits or scene collisions or minimizing energy. However, I'm not certain that enough people would use this feature that a plugin-based system is better than having people fork MoveIt themselves to implement different servoing logic.

**Additional context**
I'm in a graduate course on robot motion with @kavinda14 and we looked at DLS in `moveit_servo` for a project.
* We made a [PDF write-up](https://drive.google.com/file/d/1pjryBSiPQn6LnJT6ZxInfmbDDhm6oc_c/view) of our results
* We have cool videos of [the default arm behavior](https://youtu.be/XJkCxbw7994) and [our DLS implementation](https://youtu.be/rOcLOdW4_JA)
* We implemented DLS on a [fork of MoveIt](https://github.com/playertr/moveit/blob/f64a86f1d7ac519da4f811fabaff7d36de54cdb6/moveit_ros/moveit_servo/src/servo_calcs.cpp#L534-L554) and we can share our plotting and experiment code if desired.

[1] B. Siciliano and O. Khatib. Springer Handbook of Robotics. Berlin: Springer, 2016, pp. 227–230.

**Call to action**
What do you think? Is singularity avoidance an issue, and would DLS or similar methods be a viable solution?

danzimmerman, 2022-07-25:
Cool, is that a Reach underwater arm?

playertr, 2022-07-25:
Yep! That's a Blueprint Bravo Reach 7 with force/torque sensor installed.
If you're interested in simulating one yourself, you can follow the
directions in https://gitlab.com/rsa-manipulation-testbed/bravo_arm_sw and
poke around the parent project for its description and planning
repositories.

danzimmerman, 2022-07-25:
Sorry, I clicked on the paper and deleted my comment. Really interesting, thanks!

danzimmerman, 2022-07-25:
I've been interested in the discussion on DLS too. Looks like [moveit2/#1434](https://github.com/ros-planning/moveit2/pull/1434) is going to bring plugins to `moveit_servo` for ROS2 at least.

playertr, 2022-07-26:
Oh awesome! I wasn't aware of that PR, looks great!

If you happen to have a BPL Bravo that you want to use, feel free to use our ROSControl implementation from the prior link, as that sucked up a fair bit of dev time. Adding an open source license is one of our TODOs right now, but we can fast-track that if you're interested. 

[Video demo of Bravo ROSControl Driver](https://www.youtube.com/watch?v=hcYCcoQwGwI)