moveit/moveit #3159: Different planning results between c++ and python interface

jsbyysheng, 2022-06-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source build from master branch

### Steps to reproduce
In rivz, using motion planning plugin to plan robot move in `z+0.05` get correct results. (Have tried for 10 times)
```
[ INFO] [1654768863.379805972]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1654768863.379852136]: Using planning pipeline 'ompl'
[ INFO] [1654768863.381383438]: Planner configuration 'manipulator[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1654768863.381999212]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.382085290]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.382173744]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.382265368]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.393834397]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654768863.394000046]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.394359550]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.395009513]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.395179181]: ParallelPlan::solve(): Solution found by one or more threads in 0.013464 seconds
[ INFO] [1654768863.395600964]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.395681381]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.395912000]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.396021091]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.396571780]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.397003931]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.397546423]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1654768863.397604034]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654768863.397793616]: ParallelPlan::solve(): Solution found by one or more threads in 0.002359 seconds
[ INFO] [1654768863.398122700]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.398200117]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654768863.399551581]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654768863.399995843]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1654768863.400192984]: ParallelPlan::solve(): Solution found by one or more threads in 0.002201 seconds
[ INFO] [1654768863.403346263]: SimpleSetup: Path simplification took 0.003037 seconds and changed from 3 to 2 states

```
[image]

However, when using python interface, the results are all wired. (Have tried for many times.)
```
[ INFO] [1654769004.327905298]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1654769004.327931945]: Using planning pipeline 'ompl'
[ INFO] [1654769004.329302351]: Planner configuration 'manipulator[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1654769004.329758681]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.329808425]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.329859890]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.329941552]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.351190517]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654769004.351274890]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654769004.351531769]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1654769004.352135813]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654769004.352277536]: ParallelPlan::solve(): Solution found by one or more threads in 0.022740 seconds
[ INFO] [1654769004.352579709]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.352647052]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.352729478]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.352785743]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.354010976]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1654769004.354815474]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1654769004.355670929]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654769004.356622430]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654769004.356788503]: ParallelPlan::solve(): Solution found by one or more threads in 0.004378 seconds
[ INFO] [1654769004.357012408]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.357073448]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1654769004.358520461]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1654769004.358965464]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1654769004.359146759]: ParallelPlan::solve(): Solution found by one or more threads in 0.002230 seconds
[ INFO] [1654769004.376688942]: SimpleSetup: Path simplification took 0.004986 seconds and changed from 3 to 2 states
```
[image]

jsbyysheng, 2022-07-07:
> Hi,
> 
> I think I have a similar issue. For me planning in rviz with the planning plugin also works fine but if I use the C++ interface (similar as you did with python) it does find a path but its completely non-optimal and makes huge movements although the goal is just a small translation. Have you found a solution?

Unfortunately, I did not find a solution. Before, I reviewed the plugin code and thought this problem would not occur in c++. Ops.....

christinaionescu, 2022-07-07:
Would you mind sending me your c++ code? maybe I can find some error in my c++ code then :)

jsbyysheng, 2022-07-07:
> Would you mind sending me your c++ code? maybe I can find some error in my c++ code then :) Are you also using https://github.com/fmauch/universal_robot/tree/calibration_devel in calibration devel?

I am using `calibration devel`. I am sorry I am too busy to try the cpp code in noetic. I advise you to take a look at the official `motion planning` plugin code, it will be helpful.

christinaionescu, 2022-07-08:
@rhaschke thank you! changing this line from sample() to project() already worked for me. :) that saved me a lot of more debugging time.

v4hn, 2022-07-08:
@captain-yoshi This test clearly demonstrates the expected performance drop that you could not show in your tests. It's probably worth looking into this for the benchmark suite because this kind of bug (picking an inappropriate IK solution) has come up many times in the past.

captain-yoshi, 2022-07-08:
@v4hn I can now replicate the bug, the problem was with the goal constraint. I only setup a position constraint and thus the robot had to do large movements in the joint space ([length metric](https://github.com/captain-yoshi/moveit_benchmark_suite/blob/master/.doc/profilers/motion_planning.rst#metrics)) to reach the goal. As you can see in the picture below the IK seed (project vs sample) didn't affect the benchmark.

**5 runs of 200 trials for Planner RRTConnect versus Position Constraint PC2 versus Moveit Version 1.1.4 through 1.1.7**
[image]

But with a **pose** constraint goal with a small cartesian offset, the motion plan is affected a lot. TODO Add image once benchmark is done!

jsbyysheng, 2022-07-14:
> @rhaschke thank you! changing this line from sample() to project() already worked for me. :) that saved me a lot of more debugging time.

Can I know which line did you change? @christinaionescu  Unfortunately, I have tested the newest commit but nothing changed in python interface. @rhaschke

rhaschke, 2022-07-14:
@jsbyysheng, you can either revert the offending PR: https://github.com/ros-planning/moveit/pull/2872
Or you can apply the not-yet-merged fixup/cleanup PR: https://github.com/ros-planning/moveit/pull/3170

jsbyysheng, 2022-07-14:
> #2872

Thank you for so quick reply, it works.

jsbyysheng, 2022-07-14:
> @jsbyysheng, you can either revert the offending PR: #2872 Or you can apply the not-yet-merged fixup/cleanup PR: #3170
Thank you for so quick reply, it works.

PR #3170 (2022-07-19): Remove ConstraintSampler::project()
As pointed out in https://github.com/ros-planning/moveit/pull/3119#issuecomment-1160397187, the difference between `project()` and `sample()` is highly subtle and non-intuitive to users. Thus, it was suggested to drop `project()` and just use `sample()`, which is implemented with this PR.
Fixes #3159. Builds on #3119.
closed 2022-07-19 by v4hn