moveit/moveit #3162: getCurrentExpectedTrajectoryIndex() does not accommodate speed scaling factor for the real robot

vikashj21, 2022-06-15:
### Description

I am currently looking into the perception pipeline tutorial of MoveIt for obstacle avoidance. I noticed that there has been a function (isRemainingPathValid()) implemented in the plan_execution.cpp file to check whether any obstacles are introduced in its remaining path and ensure its validity. 
https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_ros/planning/plan_execution/src/plan_execution.cpp#L273-L274

The above function makes use of getCurrentExpectedTrajectoryIndex() to get the current position(path segment) at which the robot is at. 
https://github.com/ros-planning/moveit/blob/cce0ffe58c3f472fc5bf76b1ec364d29d2fa7252/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1594

However, when I adjusted the speed scaling factor using the UR5’s teach pendant, I observed that this function (getCurrentExpectedTrajectoryIndex()) returns that the trajectory has been completed even though the robot is only partially through its trajectory. Just to be clear, the function works perfectly when operating at 100% speed. Thus, my intuition is that the speed scaling factor may not have been accommodated into it (please correct me if I’m wrong).

Any general direction towards solving this is greatly appreciated.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Source build
* If source, which branch? master

gavanderhoorn, 2022-06-15:
>  Thus, my intuition is that the speed scaling factor may not have been accommodated into it (please correct me if I’m wrong).

no, that would be correct.

Please note: this function is a generic function.

'speed scaling' is UR-specific.

The whole of MoveIt is not aware of external systems dynamically adjusting joint limits, and this includes the trajectory execution parts.

closed 2022-06-16 by vikashj21

gavanderhoorn, 2022-06-16:
As an additional comment: there's nothing preventing anyone from updating MoveIt's trajectory execution to become aware of (expected) external changes to joint limits while executing trajectories.

That would require some work though. At the very least, the timeout calculations for trajectory execution would need to be updated dynamically.

My comment above was to confirm your observation, and to explain the context a bit.