moveit/moveit #3164: IK fails with real robot

rrapi, 2022-06-15:
Hi, as I am working with a UR5e and a gripper attached, I modified the urdf in such a way the end effector is in +0.18 on the z-axis. But doing so and having the same ur5e_moveit_config as before, I don't know if it is for this reason that the Ik makes very bad trajectories most of the time with collision or aborted messages.
In case this is the problem, I have to make another moveit configuration with the urdf changed?? 

For sending the pose target I am using this settings:
```
self.move_group.set_goal_position_tolerance(1e-5)
self.move_group.set_goal_orientation_tolerance(1e-5)

self.move_group.set_pose_target(self.goal)

self.move_group.allow_replanning(True)
self.move_group.set_planning_time(5)

self.move_group.go(wait=True)

rospy.sleep(0.5)
self.move_group.stop()
rospy.loginfo("move_group: goal sent")
self.move_group.clear_pose_targets()
```

...and the pose is a p = geometry_msgs.msg.Pose() message.

  I never had problems with the target position, but only with target orientation. For calculating the orientation I use tf.transormations.quaternion_from_euler in this way:
```
quat = quaternion_from_euler(0, np.radians(180), 0)
p.position = curr_pose.position
p.orientation = quat
```

And the planning frame is the "base_link".

But in some positions and configuration of the robot, just with a new orientation, the robot trajectory execution fails. 
I saw in gazebo, for not doing everything on a real robot as it is slower,  that it tries to readh that pose but it collides with itself and the trajectory that it tries to do is very strange.

I am using Ubuntu with ROS and Moveit to control the robot.

I tried a lot of things but did not found still nothing to solve my problem and there are remaining just a few days to complete this project.
I really hope that you will help me in some way to solve this problem and I will appreciate it a lot.

Best regards,
Franci

simonschmeisser, 2022-08-22:
It is hard to help you without knowing which ik solver you use and what modifications you made exactly, maybe upload your setup to a github repo and link here. Also try to give more details what you mean by "fails" and "very strange". Please add logs, screen casts etc

karansspk462000, 2026-08-21:
It seems like some issue with how you are representing angles and doing angle conversion. 
Not very sure as it didn't have much info here