moveit/moveit #3166: Moveit servo with custom end effector gives jerky movement on real robot (UR5e)

nihalar, 2022-06-26:
### Description

I'm testing moveit servo with the actual Ur5e robot. The robot moves smooth when providing continuous constant velocity using the original URDF provided by Universal Robots (Velocity passed to ee_link). 

The next test was to add an end effector. The end effector was added to the urdf (but not to the actual robot). Two different groups are defined in moveit. "Arm" group contains the joints of Ur5e and the end effector group which contains the links of the new end effector (All fixed links). Then I try to move the robot using moveit servo (passing velocity commands to ee_link, part of "arm" group). This time the robot movement is jerky. What Am I doing wrong? 

Please note: The end effector links are added to urdf only, the actual robot does not have those links attached.

Help would be much appreciated. Thanks :)

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04

AndyZe, 2022-06-27:
I don't know the answer but I'm curious if your computer is stressed during the testing. If you run `htop` or `top` during the test, does it show a high CPU load?

My initial guess is latency due to `tf`.

nihalar, 2022-08-01:
> I don't know the answer but I'm curious if your computer is stressed during the testing. If you run `htop` or `top` during the test, does it show a high CPU load?
> 
> My initial guess is latency due to `tf`.

I'm using a very high end PC. When I remove the mesh (stl file) from the urdf, the robot seems to move better :/ Any thoughts wh?

AndyZe, 2022-08-01:
That's a good hint. You can simplify the mesh with Meshlab. It's very easy to do. Just open the stl in Meshlab then Filters->Simplification.

You can also turn down the collision checking frequency here:  https://github.com/ros-planning/moveit/blob/a63580edd05b01d9480c333645036e5b2b222da9/moveit_ros/moveit_servo/config/ur_simulated_config.yaml#L58

nihalar, 2022-08-01:
> That's a good hint. You can simplify the mesh with Meshlab. It's very easy to do. Just open the stl in Meshlab then Filters->Simplification.
>
> You can also turn down the collision checking frequency [...]

But the problem occurs while running on the real robot. On simulation it works well. (I have also tried disabling collision which didnt help)

stevensu1838, 2023-03-07:
@nihalar ,
Hi, I am trying to use movit servo with UR5e robot. Could you please share what UR robot driver you used for your setup? It will be great if you can share the command lines as well. Thanks a lot. Cheers

stevensu1838, 2023-03-07:
@AndyZe Hi Andy,

I am trying to run moveit servo with Noetic for a UR5e.
For the UR robot package, I tried both of the following two:

git clone -b melodic-devel https://github.com/ros-industrial/universal_robot.git src/universal_robot
or git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot

Neither worked and there is no ee_link defined in the URDF files. There is only a link called flange.

Do you know which package @nihalar used? Coz he mentioned The robot moves smooth when providing continuous constant velocity using the original URDF provided by Universal Robots (Velocity passed to ee_link). Maybe I can give that package a go? Thanks a lot. Cheers

nihalar, 2023-03-08:
> @nihalar , Hi, I am trying to use movit servo with UR5e robot. Could you please share what UR robot driver you used for your setup? It will be great if you can share the command lines as well. Thanks a lot. Cheers

I used https://github.com/UniversalRobots/Universal_Robots_ROS_Driver for driver.
As for the urdf files, I downloaded it from github (Can't find/remember the exact repo now). I could send you the urdf package.

stevensu1838, 2023-03-08:
@nihalar Hi, yes, please. I'd love to give it a go. Thanks a million. My email is [stevensu1838@gmail.com](stevensu1838@gmail.com). Cheers

stevensu1838, 2023-03-13:
@nihalar Can you please share how you set it up? Really appreciate it. Thanks a lot