moveit/moveit #3171: Is there an option to check collisions in python interface

aprath1, 2022-07-08:
Is there an option to check if the robot and an object in the planning scene is in collision from the python interface?

rhaschke, 2022-07-08:
There is not _the_ python interface. So, which one do you refer to?
The python `PlanningScene` wrapper [provides this functionality](https://github.com/ros-planning/moveit/blob/605b28aaf4fe573000f3e4995c36c66d143e3885/moveit_core/planning_scene/src/pyplanning_scene.cpp#L54-L68). 
The `PlanningSceneInterface` wrapper (as the PSI itself) doesn't, but only allows [inspecting and modifying the `move_group` node's `PlanningScene`](https://github.com/ros-planning/moveit/blob/605b28aaf4fe573000f3e4995c36c66d143e3885/moveit_commander/src/moveit_commander/planning_scene_interface.py#L59).

It might be worth to add a constructor to the former one, to initialize once from the move_group's scene.

gavanderhoorn, 2022-07-08:
I believe this is a cross-post of https://github.com/mikeferguson/moveit_python/issues/32.

aprath1, 2022-07-13:
I am currently using a work around with the following c++ classes `PlanningSceneMonitor` and `LockedPlanningSceneRO`. With  this we can make collision requests to the current planning scene. 
For now I will close this issue, in case some one has a pythonic solution please feel free to comment afterwards.

closed 2022-07-13 by aprath1