moveit/moveit #3174: Getting a warning when attaching a box: "Empty quaternion found in pose message"

tgaspar, 2022-07-14:
### Description

MoveIt prints a warning when I try to attach an object to a link:
```
Empty quaternion found in pose message. Setting to neutral orientation.
```

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Binary
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
I set up a very simple robot with a Move Group from its base (`base_link`) to its tip (`gripper`). I want to attach a box (`attach_box`) to the robot's. I do so with the following code snippet:
```python
#!/usr/bin/env python3

import rospy
import moveit_commander

from geometry_msgs.msg import PoseStamped

rospy.init_node('attach_part')

gripper_name = rospy.get_param('~gripper_name', 'gripper')
part_size = rospy.get_param('~part_size', [0.1, 0.1, 0.1])

gripper_pose = PoseStamped()
gripper_pose.header.frame_id = gripper_name
gripper_pose.pose.orientation.w = 1.0

part_name = gripper_name + '_part'

scene = moveit_commander.PlanningSceneInterface()
rospy.sleep(1)
rospy.loginfo("Attaching part ...")
scene.attach_box(
    gripper_name,
    part_name,
    pose=gripper_pose,
    size=part_size)

rospy.loginfo("Done!")

```

When I do that, I get the warning mentioned above:
```
Empty quaternion found in pose message. Setting to neutral orientation.
```

### Expected behaviour
There should be no warning as I am setting the value of the scalar part of the quaternion to `1.0`.  Is there something else that I am missing? I am developing a node for both Melodic and Noetic. I haven't had this issue in Melodic but only in Noetic. Any help would be appreciated.

v4hn, 2022-07-15:
Thanks for reporting!

You found a leftover that was not adjusted to the new `pose` field in `moveit_msgs/CollisionObject`. Please see #3176 for my proposed fix.

closed 2022-07-15 by rhaschke

tgaspar, 2022-07-15:
Excellent! Thanks for resolving this so quickly! 🚀

Aadi0902, 2023-04-03:
@v4hn  Quick question, was this fixed for `planning_scene_interface.cpp` as well?

v4hn, 2023-04-04:
I'm not aware of any leftover place where this is still broken. If you see the warning and are sure you never pass in an invalid/uninitialized quaternion, please create a new issue.

PR #3176 (2022-07-15): adjust python PSI to co.pose interface
Another leftover from the migration to co.pose.

Fixes #3174.