moveit/moveit #3179: CI broken

rhaschke, 2022-07-25:
Looks like the latest python update in Ubuntu 20.04 broken CI:
https://github.com/ros-planning/moveit/runs/7507980799?check_suite_focus=true

wxmerkt, 2022-07-26:
I believe this is due to the [moveit/moveit:noetic-ci-shadow-fixed](https://hub.docker.com/layers/moveit/moveit/moveit/noetic-ci-shadow-fixed/images/sha256-82554facc7f81a6eb56fe4496bdc3036795fcbbbb89ce56cf24f6b6384c71469?context=explore) image being 3 days old and thus containing version 2.7.7 of EigenPy which was broken. v2.7.8 which fixed this has been released and is available now. As soon as the tagged image gets updated the CI issue should be resolved.

rhaschke, 2022-07-27:
Thanks for the hint! I triggered a new docker build.

closed 2022-07-27 by rhaschke