moveit/moveit #3181: Movit doesnt build when Nvidia Tensorrt Container is used

FHarhammer, 2022-07-29:
### Description

I am trying to build a docker image which is based on the NVidia Tensorrt image _nvcr.io/nvidia/tensorrt:22.04-py3_. The ROS Noetic installation works without any problems. After this I need to build the movit package from source but i always get this error 
```
usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
```

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 18.04
* Buildtype: Source 
* Branch: master

### Backtrace or Console output
```

root@b9123b6330e6:/moveit_ws# catkin build -j$(nproc) -DPYTHON_EXECUTABLE=/usr/bin/python3
--------------------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /moveit_ws
--------------------------------------------------------------------------------------------
Build Space:        [exists] /moveit_ws/build
Devel Space:        [exists] /moveit_ws/devel
Install Space:      [unused] /moveit_ws/install
Log Space:          [exists] /moveit_ws/logs
Source Space:       [exists] /moveit_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3
Additional Make Args:        -j16
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------------------
[build] Found 47 packages in 0.0 seconds.                                                                                                                                                   
[build] Package table is up to date.                                                                                                                                                        
Starting  >>> geometric_shapes                                                                                                                                                              
Starting  >>> moveit_msgs                                                                                                                                                                   
Starting  >>> moveit_resources_fanuc_description                                                                                                                                            
Starting  >>> moveit_resources_panda_description                                                                                                                                            
Starting  >>> moveit_resources_pr2_description                                                                                                                                              
Starting  >>> moveit_resources_prbt_support                                                                                                                                                 
Starting  >>> rviz_visual_tools                                                                                                                                                             
Starting  >>> srdfdom                                                                                                                                                                       
Finished  <<< moveit_resources_pr2_description                               [ 0.7 seconds ]                                                                                                
Finished  <<< moveit_resources_prbt_support                                  [ 0.7 seconds ]                                                                                                
Finished  <<< geometric_shapes                                               [ 1.0 seconds ]                                                                                                
Finished  <<< moveit_resources_fanuc_description                             [ 0.7 seconds ]                                                                                                
Starting  >>> moveit_resources_fanuc_moveit_config                                                                                                                                          
Finished  <<< moveit_resources_panda_description                             [ 1.0 seconds ]                                                                                                
Starting  >>> moveit_resources_dual_panda_moveit_config                                                                                                                                     
Starting  >>> moveit_resources_panda_moveit_config                                                                                                                                          
Finished  <<< moveit_resources_fanuc_moveit_config                           [ 0.7 seconds ]                                                                                                
Finished  <<< srdfdom                                                        [ 1.3 seconds ]                                                                                                
____________________________________________________________________________________________________________________________________________________________________________________________
Errors     << rviz_visual_tools:make /moveit_ws/logs/rviz_visual_tools/build.make.011.log                                                                                                   
In file included from /usr/include/c++/9/ext/string_conversions.h:41,
                 from /usr/include/c++/9/bits/basic_string.h:6496,
[... 86 lines cut ...]
Abandoned <<< moveit_ros_perception                                          [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_ros_robot_interaction                                   [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_ros_warehouse                                           [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_ros_benchmarks                                          [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_ros_planning_interface                                  [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_planners_chomp                                          [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_ros_visualization                                       [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_servo                                                   [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_setup_assistant                                         [ Unrelated job failed ]                                                                                       
Abandoned <<< moveit_simple_controller_manager                               [ Unrelated job failed ]                                                                                       
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                                  
[build]   Abandoned: 36 packages were abandoned.                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                     
[build] Runtime: 3.3 seconds total.
```

rhaschke, 2022-07-30:
> The ROS Noetic installation works without any problems.

I guess you are referring to an installation from Debian packages?
Actually, I'm surprised as the TensorRT image is based on Ubuntu 22.04, while ROS Noetic requires Ubuntu 20.04!

> After this I need to build the MoveIt package from source but i always get this error:
> ```
> usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
> ```

Did you install build dependencies (via `rosdep`)? `stdlib.h` should come from packages `libc6-dev` and `libstdc++-9-dev`:
```
libc6-dev:amd64: /usr/include/stdlib.h
libstdc++-9-dev:amd64: /usr/include/c++/9/stdlib.h
```
Do you have those files?

v4hn, 2022-08-01:
```
In file included from /usr/include/c++/9/ext/string_conversions.h:41,
                 from /usr/include/c++/9/bits/basic_string.h:6496,
                 from /usr/include/c++/9/string:55,
                 from /moveit_ws/src/rviz_visual_tools/src/remote_control.cpp:40:
/usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
compilation terminated.
```

This error is the compiler failing to compile ```#include <string>```. So as Robert pointed out above you have a problem with your compiler environment and not with MoveIt. Aside from that the error you cite is in [rviz_visual_tools](https://github.com/PickNikRobotics/rviz_visual_tools/blob/master/src/remote_control.cpp) and not here.

>  Actually, I'm surprised as the TensorRT image is based on Ubuntu 22.04, while ROS Noetic requires Ubuntu 20.04!

Which doesn't mean that it should be hard to compile things yourself in 22.04 or rely on external packagers there. ;)
How did you install noetic on 22.04 @FHarhammer ?

Harhammer, 2022-08-02:
Thanks a lot for the fast reply, unfortunatly i am on vacations at the moment but i will come back to the issue as soon as i am back in the office

closed 2022-08-16 by FHarhammer