moveit/moveit #3183: ComputeCartesianPAth returns fractions < 0.9 for simple trajectories.

goyalvarun02, 2022-08-02:
Hi , 

I have following queries:

a) While using Move IT , I observed object was just kept vertical down to gripper , but move It return fraction of 15% and xArm could not pick the object as we have checked if fraction>90% it should not pick , and sometime also I feel that the trajectory from this position is quite easy and can pick by robot but then I found fraction is <90% . What might be the issue , and what needs to be done for this ? How can I check trajectory and why move it is returning fraction (computeCartesianPath) which is <90% for simple locations?

b) I observe xarm_straight_plan service which is wrapper function to plan the path using move_group.computeCartesianPath()  , I can understand it fails sometime if fraction < 90% .But sometime I could not find any error logs , it just failed. May I know the reason of it?

c) xarm_exec_plan also wrapper function using which call move_group.execute(), It did not returns any string for failed service call . How can it return the string and we can know the precise error for it fail.

Thanks

simonschmeisser, 2022-08-02:
What exactly is xarm?

Also please add some more information including screenshots and ideally an minimal working example (MWE)

goyalvarun02, 2022-08-02:
Thanks for reply. 

Here are more information.
xARm has robot arm which can have DOF 5/6/7 . In my case I am using DOF : 7
https://github.com/xArm-Developer/xarm_ros (more info about xArm can be found here)

Setup is like this : 
[image]

For this movement , moveIt returns the fraction 8% as attached 
[image]

Sometime failed to call xarm_straight_plane which is internally calling computeCartesianPath
[image]

rhaschke, 2022-08-02:
If the problem arises with using xarm API, why don't you file your issue with the downstream project, i.e. at https://github.com/xArm-Developer/xarm_ros?
We don't know about the kinematics and joint limitations of this arm. A Cartesian path fraction less than 100% usually means either a collision was detected (which seems not to be the case here) or IK failed.

goyalvarun02, 2022-08-02:
Hi @rhaschke  
Thanks for the reply .

Internally xARM Api calls movegroup.ComputeCartesianPath() which returns fractions <0.9 . In this xArm does not plan any role . It is just returning the values . So the doubt was that only why for simple trajectory fraction value returns was 40% . 

I have observe xarm_straight_plan service which is wrapper function to plan the path using move_group.computeCartesianPath() , I can understand it fails sometime if fraction < 90% .But sometime I could not find any error logs , it just failed. May I know the reason of it?

 xarm_exec_plan also wrapper function using which call move_group.execute(), It did not returns any string for failed service call . How can it return the string and we can know the precise error for it fail.

rhaschke, 2022-08-02:
> Why for simple trajectory fraction value returns was 40%.

From the single image provided I cannot guess the intended trajectory. Hence, I cannot judge the expected simplicity of the motion. As said before, I suspect kinematics limitations.
* Check joint angle bounds.
* Try to perform the motion with the rviz `MotionPlanningDisplay`, which provides an interactive marker.  
  [image]
* Plan a `Cartesian Path` motion (tick the corresponding checkbox) from rviz and inspect the resulting trajectory.

> xarm_exec_plan also wrapper function using which call move_group.execute(), It did not returns any string for failed service call . How can it return the string and we can know the precise error for it fail.

This needs to be implemented downstream! MoveIt's `ComputeCartesianPath` service (and methods) [return an error code](https://github.com/ros-planning/moveit_msgs/blob/152f6f6ac1ec83e0028c806831a4fbca32008f5e/srv/GetCartesianPath.srv#L65-L66).

AndyZe, 2022-08-02:
I think chances are good that the start position of the robot is near a singularity. From your image, it looks like the robot cannot move left or right. I would try a different start position. It looks like all of the joints are nearly lying on the same plane right now -- that's not good.

goyalvarun02, 2022-08-02:
> I think chances are good that the start position of the robot is near a singularity. From your image, it looks like the robot cannot move left or right. I would try a different start position. It looks like all of the joints are nearly lying on the same plane right now -- that's not good.

I am new to ROS , I understood singularity means conditions which blocks the robot or End effector at certain conditions.  I have tested Robot was easily can move left /right/up/down even along 7 DOF.

What it means to have all joints lying nearly on the same plane ? Is it bad to have all joints in same plan or what is the impact of that ? could you please clarify.

AndyZe, 2022-08-02:
I'm going to close the issue because there's nothing we can do to help. It sounds like you ought to read up on singularities or watch some YouTube videos.

>  I have tested Robot was easily can move left /right/up/down even along 7 DOF.

Can it maintain the same end-effector orientation while it's moving, though? I doubt it.

closed 2022-08-02 by AndyZe

goyalvarun02, 2022-08-02:
> > Why for simple trajectory fraction value returns was 40%.
> 
> From the single image provided I cannot guess the intended trajectory. Hence, I cannot judge the expected simplicity of the motion. As said before, I suspect kinematics limitations.
> I am new to Robotics, So May I know on which basis we can say the issue might be due to kinematics limitations. I am trying to going more deep down to understand the problem.
> * Check joint angle bounds.
> * Try to perform the motion with the rviz `MotionPlanningDisplay`, which provides an interactive marker.
>   [image]
> * Plan a `Cartesian Path` motion (tick the corresponding checkbox) from rviz and inspect the resulting trajectory.
> I have check the bound angles too.
> > xarm_exec_plan also wrapper function using which call move_group.execute(), It did not returns any string for failed service call . How can it return the string and we can know the precise error for it fail.
> 
> This needs to be implemented downstream! MoveIt's `ComputeCartesianPath` service (and methods) [return an error code](https://github.com/ros-planning/moveit_msgs/blob/152f6f6ac1ec83e0028c806831a4fbca32008f5e/srv/GetCartesianPath.srv#L65-L66).
okay noted with thanks.

liujun0808, 2023-10-28:
Hi！ I have encountered the same problem as you recently， fraction was 40%. Do you solve it?

charleswilmot, 2024-01-17:
I believe this issue is caused because some of the waypoints on the path can be reached by multiple joint positions.

From one waypoint to the next, the plan requests the arm to perform a complete reconfiguration (ie. turn joints 180 degs around)

This is then detected by the `joint_threshold` thingy, and the planing is aborted.