moveit/moveit #3185: How does moveit load multiple planners and then select them online?

cgdsss, 2022-08-10:
I observed that moveit seems to support switching planners, such as move_group.set_planning_pipeline_id, but I don't know how to configure the move_group.launch file to load multiple planning_pipelines in one move_group. Do you have any relevant examples?

```bash
> **rosservice call /query_planner_interface**
planner_interfaces: 
  - 
    name: "OMPL"
    pipeline_id: ''
    planner_ids: 
      - arm
      - arm[BFMTkConfigDefault]
      - arm[BKPIECEkConfigDefault]
      - arm[BiESTkConfigDefault]
      - arm[BiTRRTkConfigDefault]
      - arm[ESTkConfigDefault]
      - arm[FMTkConfigDefault]
      - arm[KPIECEkConfigDefault]

```

cgdsss, 2022-08-10:
I want to use ompl, pilz_industrial_motion_planner and stomp planner at the same time and switch them online depending on the situation.

cgdsss, 2022-08-17:
[image]
It seems that the melodic version does not support it, but the noetic version does. Is this the case? When will melodic be supported?

cgdsss, 2022-08-17:
@rhaschke thanks

cgdsss, 2022-08-17:
https://github.com/ros-planning/moveit_resources/blob/master/prbt_moveit_config/launch/move_group.launch

cgdsss, 2022-08-17:
https://github.com/ros-planning/moveit_resources/blob/melodic-devel/prbt_moveit_config/launch/move_group.launch

cgdsss, 2022-08-17:
How can I use the compiled noetic version of moveit in a melodic environment?

rhaschke, 2022-09-12:
Sorry for the late reply. Examples can be found in `moveit_resources` - as you did.

> It seems that the melodic version does not support it, but the noetic version does. Is this the case? 

That's correct.

> When will melodic be supported?

Probably never. There are only very few maintainers left for ROS1 and we focus on Noetic. You can build the master branch in a Melodic environment without problems. So, if you need this feature, that's the way to go (or update to Noetic).

closed 2022-09-12 by rhaschke