moveit/moveit #3186: Calling `moveit_commander` function `set_joint_value_target` crashes without warnings if `len(position) < len(name)`

FSund, 2022-08-16:
### Description

Calling `set_joint_value_target` ([link to source](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_commander/src/moveit_commander/move_group.py#L199)) with `arg1` of type `JointState` and `len(arg1.name) > len(arg1.position)` leads to moveit_commander crashing without any error messages (except `process has died` -- the logs contain nothing of interest).

Is this the expected outcome, or should the length of these fields be verified somewhere?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Latest version from apt

### Steps to reproduce
```
#!/usr/bin/env python
import rospy
import moveit_commander
from sensor_msgs.msg import JointState

if __name__ == "__main__":
    rospy.init_node("test_moveit_commander")

    move_group = moveit_commander.MoveGroupCommander(
        name="shoulders",  # change this to fit whatever robot you are using
        robot_description="robot_description",
    )

    joints = move_group.get_active_joints()
    js = JointState()
    js.name = [joints[0]]
    js.position = []

    move_group.set_joint_value_target(js)
```

rhaschke, 2022-08-16:
It is definitely asserted that these arrays have same length. Please file a PR adding a corresponding python assertion. Thanks.

FSund, 2022-08-16:
@rhaschke I've added a draft PR with a suggestion of how to fix the issue, with some questions in a comment.

PR #3187 (2022-08-16): Test `moveit_commander.set_joint_value_target` with JointState argument
### Description

- Add tests of `moveit_commander.set_joint_value_target` with `JointState` argument
- Add test of `moveit_commander.set_joint_value_target` with incomplete `JointState` argument

Fixes #3186 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](htt [... cut]
closed 2022-08-16 by v4hn