moveit/moveit #3189: Allow to stream "stopless" trajectories executions

Hugal31, 2022-08-18:
**Is your feature request related to a problem? Please describe.**

I want to execute a very long trajectory (multiple minutes). However, to avoid wasting time computing such a long trajectory, I compute a small part, execute it, and compute the next part while executing. I already implemented that. However, I have to stop between each trajectories because the current execution manager doesn't allow to schedule the next one.

joint_trajectory_controller allow [trajectory replacement](https://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement), but MoveIt! doesn't allow to use this feature.

**Describe the solution you'd like**

Some way to schedule a trajectory at a specific date, during a trajectory execution, that allows to take advantage of the trajectory replacement implemented by the joint_trajectory_controller so the movement is never interrupted.

**Describe alternatives you've considered**

Because this is maybe a bit too specific, an alternative would be to let the user implement the trajectory execution manager as a plugin.

simonschmeisser, 2022-08-22:
We discussed implementing a mode like this in #3156 and in the discourse channel, maybe you would like to add some input.

Do your trajectory segments start and end with zero velocity or with the same continuous non-zero velocity? Would you expect the TrajectoryExecutionManager to adjust the velocities and accelerations?

Hugal31, 2022-08-22:
We would like to send trajectory segment with continuous non-zero velocities. Ideally, the TrajectoryExecutionManager would not have to adjust anything, as this is already implemented in the follow_joint_trajectory with the trajectory replacement mechanism.

We could imagine mode in thich the TEM would send the JointTrajectories to the controllers without explicitly canceling the previous goal (which stops the motion in the joint_trajectory_controller).

Exemple:

Current behavior: 
```
Client                    MoveIt!                   Controllers
   |                         |                           |
   |  -- Trajectory 1 -->    |                           |
   |                         |    -- Trajectories 1 -->  |
   |                         |                           |
   |  -- Trajectory 2 -->    |                           |
   |                         |       -- Cancel 1->       |
   |   <- 1 Canceled --      |                           |
   |                         |    -- Trajectories 2 -->  |
```
Desired behavior
```
Client                    MoveIt!                   Controllers
   |                         |                           |
   |  -- Trajectory 1 -->    |                           |
   |                         |    -- Trajectories 1 -->  |
   |                         |                           |
   |  -- Trajectory 2 -->    |                           |
   |                         |    -- Trajectories 2 -->  |
   |                         |                           |
```
with maybe a `Client  <-- Trajectory 1 canceled -- MoveIt`.

Also, where can I find this specific Discourse topic?

gavanderhoorn, 2022-08-22:
@Hugal31 what would be the advantage of going through the TEM? As you write yourself, the `JointTrajectoryController` already supports trajectory replacement.

Hugal31, 2022-08-22:
Fair point. But having two APIs to execute trajectories + two places where to get the ControllerHandles is not ideal.

simonschmeisser, 2022-08-22:
The discussion is going on in Discord, sorry I tend to mix up the names ...

Asynchronously pushing several trajectories (or segments) is already supported in prototypes

v4hn, 2022-08-23:
On Mon, Aug 22, 2022 at 05:40:47AM -0700, G.A. vd. Hoorn wrote:
> what would be the advantage of going through the TEM? As you write yourself, the `JointTrajectoryController` already supports trajectory replacement.

Two potential advantages I see are
- TEM supports multiple FJT controllers in one interface, so you don't have to split it manually if you have multiple pieces of hardware that are not in one ros_control loop
- The new version of TEM might support collision checking for trajectories, so it could abort/reject trajectories based on the changing planning scene. That's not working with the current version though.

These are not compelling reasons for many scenarios of course.

Hugal31, 2022-08-23:
> aren't they both action servers?

Yes, but not the same actions type (`moveit_msgs/ExecuteTrajectoryAction` vs `FollowJointTrajectoryAction`) and not the same. Also the topics are not the same.

> two places where to get the ControllerHandles
> ?

The TEM already does the work of knowing which `MoveItControllerHandle` to instantiate and how to dispatch a trajectory to multiple controllers, as v4hn pointed.

> Asynchronously pushing several trajectories (or segments) is already supported in prototypes

Do you have a link to one of those prototype @simonschmeisser?

cambel, 2022-08-25:
@Hugal31 the prototypes are:
First version: https://github.com/ros-planning/moveit/pull/3169
Second version:  https://github.com/cambel/moveit/pull/2
Latest version: https://github.com/cambel/moveit/pull/3 (WIP)