moveit/moveit #3193: Planning Interface is failing in Melodic Buildfarm

vatanaksoytezer, 2022-08-23:
`moveit_ros_planning_interface` has been failing in the buildfarm since August 15: https://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_ros_planning_interface__ubuntu_bionic_amd64__binary/146/console

We should fix and release for Melodic.

The error is as follows:

```
23:20:00 	cd obj-x86_64-linux-gnu && tail -v -n \+0 CMakeFiles/CMakeError.log
23:20:00 ==> CMakeFiles/CMakeError.log <==
23:20:00 Determining if the pthread_create exist failed with the following output:
23:20:00 Change Dir: /tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp
23:20:00 
23:20:00 Run Build Command:"/usr/bin/make" "cmTC_b1d28/fast"
23:20:00 make[2]: Entering directory '/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
23:20:00 /usr/bin/make -f CMakeFiles/cmTC_b1d28.dir/build.make CMakeFiles/cmTC_b1d28.dir/build
23:20:00 make[3]: Entering directory '/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
23:20:00 Building C object CMakeFiles/cmTC_b1d28.dir/CheckSymbolExists.c.o
23:20:00 /usr/lib/ccache/cc   -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2    -o CMakeFiles/cmTC_b1d28.dir/CheckSymbolExists.c.o   -c /tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/CheckSymbolExists.c
23:20:00 Linking C executable cmTC_b1d28
23:20:00 /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_b1d28.dir/link.txt --verbose=1
23:20:00 /usr/lib/ccache/cc -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2     -rdynamic CMakeFiles/cmTC_b1d28.dir/CheckSymbolExists.c.o  -o cmTC_b1d28 
23:20:00 CMakeFiles/cmTC_b1d28.dir/CheckSymbolExists.c.o: In function `main':
23:20:00 ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/CheckSymbolExists.c:8: undefined reference to `pthread_create'
23:20:00 collect2: error: ld returned 1 exit status
23:20:00 CMakeFiles/cmTC_b1d28.dir/build.make:97: recipe for target 'cmTC_b1d28' failed
23:20:00 make[3]: *** [cmTC_b1d28] Error 1
23:20:00 make[3]: Leaving directory '/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
23:20:00 Makefile:126: recipe for target 'cmTC_b1d28/fast' failed
23:20:00 make[2]: *** [cmTC_b1d28/fast] Error 2
23:20:00 make[2]: Leaving directory '/tmp/binarydeb/ros-melodic-moveit-ros-planning-interface-1.0.10/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
23:20:00 
```

rhaschke, 2022-08-23:
Duplicate of https://github.com/ros-gbp/moveit-release/issues/10. @wxmerkt, any updates on this?

rhaschke, 2022-08-23:
There is some discussion how to resolve this issue in https://github.com/stack-of-tasks/eigenpy/pull/307.

wxmerkt, 2022-08-23:
Thanks for the report and apologies for the issue. We are working on a fix - if urgent we can revert EigenPy for Melodic

rhaschke, 2022-08-25:
Yes, [it's fixed](https://build.ros.org/job/Mbin_uB64__moveit_ros_planning_interface__ubuntu_bionic_amd64__binary/). Thanks!

closed 2022-08-25 by rhaschke

christinaionescu, 2022-10-04:
Hi I have a smiliar error:

```
CMake Error at /home/demosens/Demosens_Demonstrator/demosens_ws/src/moveit/moveit_ros/planning_interface/move_group_interface/CMakeLists.txt:8 (add_library):
  Target "moveit_move_group_interface_python" links to target
  "Boost::python38" but the target was not found.  Perhaps a find_package()
  call is missing for an IMPORTED target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
cd /home/demosens/Demosens_Demonstrator/demosens_ws/build/moveit_ros_planning_interface; catkin build --get-env moveit_ros_planning_interface | catkin env -si  /usr/bin/cmake /home/demosens/Demosens_Demonstrator/demosens_ws/src/moveit/moveit_ros/planning_interface --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/demosens/Demosens_Demonstrator/demosens_ws/devel/.private/moveit_ros_planning_interface -DCMAKE_INSTALL_PREFIX=/home/demosens/Demosens_Demonstrator/demosens_ws/install -DPYTHON_EXECUTABLE=/home/demosens/anaconda3/envs/vision_39/bin/python3.9; cd -
```
Do you have any idea?
Thank you!

wxmerkt, 2022-10-04:
Hi @christinaionescu,
Which version of Eigenpy & MoveIt do you have and are you using regular ROS or ROS-testing for your binaries?