moveit/moveit #3195: Why does group.computeCartesianPath create optimal straight line paths, my group.plan does not?

ericdusel77, 2022-08-25:
I am using ROS Noetic with MoveIt and a kinova jaco arm IRL and in simulation. When I am trying to plan I am getting drastically different solutions using two methods suggested by the move group interface tutorial. I am wondering why one is so much better than the other, and if there are some configuration steps I may be missing that will make them both be more optimal.

I have written the following code which gives me two options of planning to a certain point. One using the computeCartesianPath, and one that uses the traditional planning method from the tutorial.

```
    if (compute_cart){
        // CREATE TRAJECTORY
        std::vector<geometry_msgs::Pose> waypoints;
        waypoints.push_back(approach_pose);  // APPROACH
        waypoints.push_back(pose);  // GOAL (BUTTON)
        waypoints.push_back(approach_pose);  // APPROACH
        waypoints.push_back(home_pose);  // HOME
        moveit_msgs::RobotTrajectory trajectory;
        double fraction = group_->computeCartesianPath(waypoints,
                                                        0.01,  // eef_step - path to be interpolated at this resolution
                                                        10.0,   // jump_threshold
                                                        trajectory);

        // EXECUTE TRAJECTORY IF FULL PATH IS FOUND
        if (fraction == 1){
            group_->execute(trajectory);
        }

    } else {
        group_->setPoseTarget(approach_pose);
        evaluate_plan(*group_);
        group_->setPoseTarget(pose);
        evaluate_plan(*group_);
        group_->setPoseTarget(approach_pose);
        evaluate_plan(*group_);
        group_->setJointValueTarget(home);
        evaluate_plan(*group_);
    }

void evaluate_plan(moveit::planning_interface::MoveGroupInterface &group)
{
    moveit::planning_interface::MoveGroupInterface::Plan my_plan;
    bool result = (group.plan(my_plan) == moveit_msgs::MoveItErrorCodes::SUCCESS);
    if (result)
    {
        group.execute(my_plan);
    }
    ros::WallDuration(1.0).sleep();
}

```
However, as shown in the simulation gifs below, only the first method (computeCartesianPath) actually finds a good straight line path. 
[image]
**THIS WORKS GREAT**

Attempting to follow the same four waypoints using the plan and execute commands yields the following results. 
[image]
**THIS IS NOT A STRAIGHT LINE PATH**

Is anyone able to help me to determine what I am missing here in order to make the second option plan more closely to the first option?

rhaschke, 2022-08-25:
The two approaches are fundamentally different: `computeCartesianPath()` just performs Cartesian interpolation with _collision checking_ (but no planning), `plan()` performs sampling-based planning and _obstacle avoidance_ (resulting in random trajectories).
Have a look [here](https://picknik.ai/cartesian%20planners/moveit/motion%20planning/2021/01/07/guide-to-cartesian-planners-in-moveit.html) for an overview of these and alternative approaches.
You need to answer the question how to handle obstacles in the straight-line Cartesian path? Do you want to move around them and deviate from the Cartesian path?

ericdusel77, 2022-08-25:
Oh, that makes so much sense.  Thank you for the response!

closed 2022-08-25 by ericdusel77