moveit/moveit #3199: PlanningScene python interface

gabrielpeixoto-cvai, 2022-08-31:
Hello all,

I have been using moveit for my research (Python). I want to check for collision and know that moveit_commander (RobotCommander, PlanningSceneInterface) does not support CheckCollision or CheckSelfCollision methods. On a [previous thread](https://github.com/ros-planning/moveit/issues/3171)  people pointed to the [PlanningScene python bind](https://github.com/ros-planning/moveit/blob/605b28aaf4fe573000f3e4995c36c66d143e3885/moveit_core/planning_scene/src/pyplanning_scene.cpp#L54-L68). However, I could not import this from my environment with moveit.

My question is, should I rebuild moveit from sources to access these binds? if not how should I access these binds from python by installing moveit from packages? Currently I am using noetic moveit docker.

Furthermore, I also found [this "workaround"](https://answers.ros.org/question/203633/collision-detection-in-python/) that enables checking state validity in python using services but is this equivalent to the PlanningScene methods?

Can someone clarify these points for me?

Thanks in advance for your attention.

rhaschke, 2022-09-02:
> people pointed to the [PlanningScene python bind](https://github.com/ros-planning/moveit/blob/605b28aaf4fe573000f3e4995c36c66d143e3885/moveit_core/planning_scene/src/pyplanning_scene.cpp#L54-L68). However, I could not import this from my environment with moveit.

You should be able to import `PlanningScene` like this:
`from moveit.core.planning_scene import PlanningScene`

> I also found [this "workaround"](https://answers.ros.org/question/203633/collision-detection-in-python/) that enables checking state validity in python using services but is this equivalent to the PlanningScene methods?

Yes. The `GetStateValidity` service just calls the underlying collision checking routine:
https://github.com/ros-planning/moveit/blob/7949bd6ea776e1268b8ff9e2015cad5cb695b829/moveit_ros/move_group/src/default_capabilities/state_validation_service_capability.cpp#L75

gabrielpeixoto-cvai, 2022-09-10:
Hello @rhaschke,

First I would like to thank you for the quick response. I was finishing other things and did not have time to investigate this topic last week, but now I did some study on this topic. Indeed I could import the PlanningScene class using `from moveit.core.planning_scene import PlanningScene` but I could not instantiate it because it requires `pymoveit_core.planning_scene.PlanningScene(robot_model: pymoveit_core.robot_model.RobotModel, world: pymoveit_core.collision_detection.World = <pymoveit_core.collision_detection.World object at 0x7f59ff4361f0>)
` and I tried to instantiate `from moveit.core.robot_model import RobotModel` but this object requires `pymoveit_core.robot_model.RobotModel(urdf_model: urdf::ModelInterface, srdf_model: srdf::Model)
` and though I have the path to the URDF file I do not have the SRDF model (sorry for my lack of understanding). Additionally, the [state validity C++ method](https://github.com/ros-planning/moveit/blob/7949bd6ea776e1268b8ff9e2015cad5cb695b829/moveit_ros/move_group/src/default_capabilities/state_validation_service_capability.cpp#L55) obtain the robot model from the **planning_scene_monitor**, is there a way to get the same behavior in python? I tried to search bu moveit.core has no planning_scene_monitor:
```
>>> dir(moveit)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'core']
>>> dir(moveit.core)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'collision_detection', 'kinematic_constraints', 'load_robot_model', 'planning_scene', 'robot_model', 'robot_state', 'transforms']
>>> dir(moveit.core.collision_detection)
['AllowedCollisionMatrix', 'BodyType', 'CollisionRequest', 'CollisionResult', 'Contact', 'ROBOT_ATTACHED', 'ROBOT_LINK', 'WORLD_OBJECT', 'World', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
>>> dir(moveit.core.kinematic_constraints)
['ConstraintEvaluationResult', 'KinematicConstraintSet', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'constructGoalConstraints']
>>> dir(moveit.core.load_robot_model)
['__call__', '__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__name__', '__ne__', '__new__', '__qualname__', '__reduce__', '__reduce_ex__', '__repr__', '__self__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__text_signature__']
>>> dir(moveit.core.planning_scene)
['PlanningScene', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
>>> dir(moveit.core.robot_model)
['JointModelGroup', 'RobotModel', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
>>> dir(moveit.core.robot_state)
['RobotState', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'jointStateToRobotState', 'robotStateToRobotStateMsg']
>>> dir(moveit.core.transforms)
['Transforms', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
```
Additionally, I also tried to search for [messages](https://github.com/ros-planning/moveit_msgs/tree/master/msg) or [services](https://github.com/ros-planning/moveit_msgs/tree/master/srv) that could give me this information, but I could not find any. The documentation is not that detailed also.

rhaschke, 2022-09-10:
If the state validity service is a working solution for you, please consider augmenting the MoveIt python tutorial. Thanks.

gabrielpeixoto-cvai, 2022-09-10:
For now, it seems to fulfill my requirements. I guess that if I want finer control I should use C++.
Is there any guideline on how to contribute to the MoveIt python tutorials? I would like to contribute with the information I found.

rhaschke, 2022-09-10:
Great that you want to contribute. See [here](https://moveit.ros.org/documentation/contributing/) for instructions on how to do so. Closing this issue as resolved.

closed 2022-09-10 by rhaschke

simonschmeisser, 2022-09-10:
Just a side note about `PlanningSceneMonitor`, it is not part of the "ROS independent" MoveIt Core but in the moveit.ros "module". I'm not a user of MoveIts Python API so I don't know how this translates there but maybe you can find it there.

pclay614, 2023-04-04:
Noting that there is a function exposed (`load_robot_model`) that you can pass in the URDF and SRDF paths (as strings) to get the RobotModel.

`from pymoveit_core import load_robot_model` (to get function into the namespace)