moveit/moveit #3206: Link 'panda_link(0-7)_sc' and 'panda_hand_sc' are not known to URDF. Cannot disable/enable collisons

WangHonghui123, 2022-09-04:
### Description

When I try to input "roslaunch panda_moveit_config demo.launch" in the terminal, it shows:
[ WARN] [1662286780.413483924]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.413686723]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.413777536]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.413939808]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.414784335]: Link 'panda_link4_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.414912294]: Link 'panda_link5_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.415584525]: Link 'panda_link6_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.415716295]: Link 'panda_link7_sc' is not known to URDF. Cannot disable/enable collisons.
....
[ WARN] [1662286780.422768697]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1662286780.422782740]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
.....

When I use "Plan" Command to run, the error happens, shown as follows:

[image]

I would like to know how to fix it. Thanks.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source Build
* Moveit 1 - Noetic

rhaschke, 2022-09-04:
Please update [franka_description](https://github.com/frankaemika/franka_ros/tree/develop/franka_description) to version 0.9.1 - either using the [`ros-testing` repo](http://wiki.ros.org/TestingRepository) or building it from source. This package was released into Noetic, but not yet synced.

WangHonghui123, 2022-09-04:
Thank you for your help. It can work.

WangHonghui123, 2022-09-04:
Thank you for your help. @rhaschke. It is able to work.

closed 2022-09-04 by rhaschke

OoooL19, 2022-09-04:
Hi, I have the same issue when I use Moveit1 - melodic. Do you know what can I fix this issue?
Thank you

rhaschke, 2022-09-04:
For Melodic you need to use franka_description from source. Unfortunately, its Melodic release was reverted.

OoooL19, 2022-09-04:
> For Melodic you need to use franka_description from source. Unfortunately, its Melodic release was reverted.

Thanks. it works now.