moveit/moveit #3208: Not allowed to create a plan for a single joint within a move group without moving all joints

jeferrandiz98, 2022-09-06:
### Description

When using the [MoveGroupCommander](http://docs.ros.org/en/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html) and planning a trajectory from the current state to a target value of a single joint through [set_joint_value_target](http://docs.ros.org/en/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a55db2d061bbf73d05b9a06df7f31ea39) and using a dictionary as an argument, if this dictionary contains fewer joints than the move group, when planning, the trajectory of the remaining joints will go from their current state to 0 (which could lead to unexpected movements) instead of not moving these joints at all.

I understand that the documentation recommends [here](http://docs.ros.org/en/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#a8280f121bc31a59de9f07c94e957f1be) to reset the target values of the joints by doing **setJointValueTarget(getCurrentJointValues())** and then overwrite the target of the joints you would like to modify, but a trajectory would be planned from the current state to the current state, which could vary depending on sensor noise.

Is it actually possible to create a plan within a move group that does not move at all the joints whose target is not specified? If it is not, should this be a new feature?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04.4
* Binary build - 1.1.9 Noetic-release

### Steps to reproduce

```
from moveit_commander import MoveGroupCommander

# Right hand is a group that contains several joints
move_group_commander = MoveGroupCommander("right_hand")

move_group_commander.set_start_state_to_current_state()

# Setting target only for one joint within the move group
move_group_commander.set_joint_value_target({"rh_FFJ4": 0.1})

plan = move_group_commander.plan()[1]
```

### Expected behaviour
A plan (RobotTrajectory) should be created only for joint rh_FFJ4.

### Actual behaviour
The plan contains a trajectory for all the joints in the move group, which go from the current state to 0, except joint rh_FFJ4 that goes from the current state to the specified target (0.1).

beatrizleon, 2022-09-06:
We want to use the fuctionallity of partial joint trajectories implemented a while back in the [joint_trajectory_controller](https://github.com/ros-controls/ros_controllers/tree/noetic-devel/joint_trajectory_controller). If we add the argument "allow_partial_joints_goal: true" to the definition of the controller, then partial trajectories can be sent and executed without affecting other joints. Here is an example of that definition: [link](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/joint_trajectory_controller/test/rrbot_partial_controllers.yaml).

rhaschke, 2022-09-06:
I agree that just providing a few joints via `set_joint_value_target()` should maintain the other joints. This bug should be fixed.
However, this does not imply that these other joints may not participate in planning: all joints of the joint model group are allowed to move! Thus, `allow_partial_joints_goal` will not work - apart from the fact that this feature is not supported by MoveIt.

rhaschke, 2022-09-07:
Initializing the goal state with current joint values as documented, solves the stated problem:
```
move_group_commander.set_start_state_to_current_state()
move_group_commander.set_joint_value_target(move_group_commander.get_current_joint_values())
move_group_commander.set_joint_value_target({"panda_joint2": 0.1})
```
The target state maintained by the MGI is initialized with default values (centered in the joint ranges):
https://github.com/ros-planning/moveit/blob/0a0fc562b79984e497b9e39d3ff554705032cbbb/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L121
The `set_joint_value_target()` methods just modify this default state.

If you want to move a single joint only, you have to define a separate move group for it.

closed 2022-09-07 by rhaschke