moveit/moveit #3209: Can achieve move request

arnauddomes, 2022-09-06:
On some trajectory , the planner only achieve few purcent of the trajectory:

```
[ INFO] [1662454965.319077141]: Received request to compute Cartesian path
[ INFO] [1662454965.319391330]: Attempting to follow 1 waypoints for link 'panda_link8' using a step of 0.001000 m and jump threshold 10.000000 (in global reference frame)
[ INFO] [1662454965.376369288]: Computed Cartesian path with 20 points (followed 31.666667% of requested trajectory)
[ INFO] [1662454965.377433531]: Execution request received
[ INFO] [1662454965.884138345]: Controller 'position_joint_trajectory_controller' successfully finished
[ INFO] [1662454965.983672836]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1662454965.983748887]: Execution completed: SUCCEEDED

```

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04

Here is the code I use to generate trajectory:

 ```
def set_robot_position_with_path(self, pose):
        print("set_robot_position_with_path {}".format(pose))
        pose_goal = geometry_msgs.msg.Pose()

        pose_goal.position.x = pose["position"][0]
        pose_goal.position.y = pose["position"][1]
        pose_goal.position.z = pose["position"][2]

        pose_goal.orientation.x = pose["orientation"][0]
        pose_goal.orientation.y = pose["orientation"][1]
        pose_goal.orientation.z = pose["orientation"][2]
        pose_goal.orientation.w = pose["orientation"][3]

        self.move_group.set_start_state_to_current_state()

        waypoints = []
        wpose = self.move_group.get_current_pose().pose
        wpose.position.x = pose["position"][0]
        wpose.position.y = pose["position"][1]
        wpose.position.z = pose["position"][2]
        wpose.orientation.x = pose["orientation"][0]
        wpose.orientation.y = pose["orientation"][1]
        wpose.orientation.z = pose["orientation"][2]
        wpose.orientation.w = pose["orientation"][3]
        waypoints.append(copy.deepcopy(wpose))
        (plan, fraction) = self.move_group.compute_cartesian_path(
            waypoints,  # waypoints to follow
            0.001,  # eef_step
            10)  # jump_threshold

        velocity_command = 0.1
        plan = self.move_group.retime_trajectory(moveit_commander.RobotCommander().get_current_state(),
                                                 plan,
                                                 velocity_command)
        ret = self.execute_plan(plan)

```

rhaschke, 2022-09-07:
MoveIt distinguishes straight-line Cartesian path generation (`compute_cartesian_path` used above) and sampling-based trajectory planning (plan()). See the [python tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html).
rviz uses the latter by default. You need to explicitly tick the checkbox "Use Cartesian path" for the former.
Cartesian path planning might run into robot singularities, which will cause the motion generator to not finish its path.

closed 2022-09-07 by rhaschke

arnauddomes, 2022-09-07:
@rhaschke Thnak you for the answer.

I modify the program to use trajectory planning. Here is the result:

https://streamable.com/v7l0nk

At the beginning of the video, the generate trajectory comes from my program and the second trajectory come from rviz. 
How can I steup my trajectory planning to get the same plannification?

rhaschke, 2022-09-07:
Please provide your program if you want me to help.

arnauddomes, 2022-09-07:
Here is an example code I use.

I want to setup a basic code to be able to add object in the scene and the to be able to generate trajectory with avoid collision:

```

def main():
        joints_position = [........]

        robot = moveit_commander.RobotCommander()
        scene = moveit_commander.PlanningSceneInterface()

        rospy.sleep(2)
        while not rospy.search_param('robot_description_semantic') and not rospy.is_shutdown():
            time.sleep(0.5)

        group_name = "panda_arm"
        move_group = moveit_commander.MoveGroupCommander(group_name)
        move_group.set_max_velocity_scaling_factor(0.2)
        move_group.set_max_acceleration_scaling_factor(0.1)

        display_trajectory_publisher = rospy.Publisher('/move_group/display_planned_path',
                                                            moveit_msgs.msg.DisplayTrajectory,
                                                            queue_size=20)

        planning_frame = move_group.get_planning_frame()
        group_names = robot.get_group_names()

        print("set_robot_joints_value joints_position {}".format(joints_position))
        joint_goal = move_group.get_current_joint_values()
        print("joint_goal {}".format(joint_goal))
        joint_goal[0] = joints_position[0]
        joint_goal[1] = joints_position[1]
        joint_goal[2] = joints_position[2]
        joint_goal[3] = joints_position[3]
        joint_goal[4] = joints_position[4]
        joint_goal[5] = joints_position[5]
        joint_goal[6] = joints_position[6]
        move_group.set_joint_value_target(joint_goal)
        plan = move_group.plan()
        move_group.go(wait=True)
        move_group.stop()
```

rhaschke, 2022-09-07:
For me, this example works as expected. I guess you simply specify a weird joint target position. Unfortunately, you omitted that crucial part. You can also pass the `joints_positions` array directly:
`move_group.set_joint_value_target(joints_position)`

Here are some other alternatives:
```python
import moveit_commander

robot = moveit_commander.RobotCommander()
scene = moveit_commander.PlanningSceneInterface()

group_name = "panda_arm"
move_group = moveit_commander.MoveGroupCommander(group_name)
move_group.set_max_velocity_scaling_factor(0.2)
move_group.set_max_acceleration_scaling_factor(0.1)

def plan_and_execute():
	success, trajectory, time, error_code = move_group.plan()
	return success and move_group.execute(trajectory, wait=True)

def go():
	return move_group.go(wait=True)

i = 0
while True:
	# set the next target:
	if i % 2 == 0:  # either from named pose
		move_group.set_named_target("ready")
	else:  # or from joint dictionary
		move_group.set_joint_value_target(dict(panda_joint2=0.1))
	i += 1
	# plan and execute motions, both variants are identical
	#success = plan_and_execute()
	success = go()

```

arnauddomes, 2022-09-12:
@rhaschke Thank you it works properly.