moveit/moveit #3210: Octomap detect arm as obstacle

adamanov, 2022-09-07:
Hello, 

I am performing a simple pick and placement pipeline like in tutorial, for a detecting, and picking the box.
The box is added to the planning scene to be able to deactivate collision checking between the gripper and the box as expected. 

And as you can see, the box is also removed from octomap, after that perform execution of caclulated trajectory. 

[image]

However, when the robot arm apper into camera view, the octamap detected the arm as obstacle, and the arm stops. 

I tried both plugings `DepthImageOctomapUpdater` and `PointCloudOctomapUpdater` but the results is same. 

```
sensors:
  # - sensor_plugin: occupancy_map_monitor/DepthImageOctomapUpdater
  #   image_topic: /jackal/front_realsense/depth/image_rect_raw
  #   queue_size: 1                      # The number of images to queue up.
  #   near_clipping_plane_distance: 0.01 # The minimum distance before lack of visibility. (meters)
  #   far_clipping_plane_distance: 1.0   # The maximum distance before lack of visibility.
  #   shadow_threshold: 0.2              # The minimum brightness of the shadow map below an entity for its dynamic shadow to be visible
  #   padding_scale: 5.0                 # Should always be >= 1.0. Scale up collision shapes in the scene before excluding them from the octomap. 
  #   padding_offset: 0.03               # Absolute padding (in m) around scaled collision shapes when excluding them from the octomap.
  #   max_update_rate: 1.0
  #   point_subsample: 122
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /jackal/front_realsense/depth/color/points
    max_range: 0.7
    point_subsample: 122
    padding_offset: 0.03
    padding_scale: 1.1
    max_update_rate: 5.0
    filtered_cloud_topic: filtered_realsense_cloud
````
How can I remove the arm as collision object? I guess self collision does not work properly.

simonschmeisser, 2022-09-10:
It seems to me that most of the gripper has been cut out from the point cloud as there are only very few filled octomap voxels? Try visualizing the `filtered_realsense_cloud` to see how much has been cut out. Also the point cloud in the screenshot does not match the gripper very well. You might want to increase `padding_offset` or `padding_scale` and also improve your hand-eye calibration.

Roger-Sh, 2022-10-18:
Have you tried to increase padding_offset?

adamanov, 2022-11-16:
Sorry for the late response, I was on holiday. We figured out that the URDF model of the gripper did not accurately represent the real robot gripper. Therefore, the cloud was not correctly cut out from the scene, since the detected gripper cloud was not matched with the urdf model.

closed 2022-11-16 by adamanov