moveit/moveit #3212: ImportError: libmoveit_py_bindings_tools.so.1.1.9: cannot open shared object file: No such file or directory

elfinsirius, 2022-09-09:
I have a computer connected to a robot acting as a server with Apache2. The user on the site creates a program in Blockly, the code of this program is executed on the server computer. Everything was fine until I added MoveGroupCommander, because an error now appears and nothing is working.

`Traceback (most recent call last):
File "/var/www/robot/static/code.py", line 14, in
from robot_api import Robot
File "/var/www/robot/static/robot_api.py", line 8, in
from moveit_commander.move_group import MoveGroupCommander
File "/opt/ros/noetic/lib/python3/dist-packages/moveit_commander/init.py", line 2, in
from .roscpp_initializer import *
File "/opt/ros/noetic/lib/python3/dist-packages/moveit_commander/roscpp_initializer.py", line 35, in
from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: libmoveit_py_bindings_tools.so.1.1.9: cannot open shared object file: No such file or directory`

Here is the output of `find / -name 'libmoveit_py_bindings_tools.so*' -print`
`/opt/ros/noetic/lib/libmoveit_py_bindings_tools.so`
`/opt/ros/noetic/lib/libmoveit_py_bindings_tools.so.1.1.9`
`/usr/lib/apache2/modules/libmoveit_py_bindings_tools.so.1.1.9`
`find: ‘/run/user/1000/doc’: Permission denied`
`find: ‘/run/user/1000/gvfs’: Permission denied`

How can i fix that?

rhaschke, 2022-09-09:
> `/opt/ros/noetic/lib/libmoveit_py_bindings_tools.so.1.1.9`
> `/usr/lib/apache2/modules/libmoveit_py_bindings_tools.so.1.1.9`

Why do you have the lib both in /opt/ros (the correct location) and /usr/lib/apache2 (wrong)?
Are those files readable by the apache user? Looks like you screwed your environment.

Can you perform `from moveit_commander.move_group import MoveGroupCommander` in a standard ROS context?

elfinsirius, 2022-09-09:
I tried to solve this problem by transferring file into `/usr/lib/apache2/modules` but id didnt work.
Yes, these file are readable by www-data.
If i create python file and try this `from moveit_commander.move_group import MoveGroupCommander` it works fine.

closed 2022-09-09 by rhaschke