moveit/moveit #3213: Compute cartesian path jointspace jumps

thomasw2, 2022-09-09:
With regards to issue #797, we have reproduced this jointspace jump with KDL kinematics plugin behavior. attached is a video showing the jointspace jump we saw, and the corresponding script that produces it. the script is a cut down and modified version of move_group_python_interface_tutorial. We used the default panda_moveit_config from here(noetic-devel branch):
https://github.com/ros-planning/panda_moveit_config.git
This branch has the KDL kinematics plugin configured in its kinematics.yaml config by default. 

Note: we originally saw this issue on a ur3e arm.

----------------------------------------
@thomasw2, please provide a minimal example (ideally using panda_moveit_config) reproducing your issue - in a new issue.
It's hard to help without any concrete info.

_Originally posted by @rhaschke in https://github.com/ros-planning/moveit/issues/797#issuecomment-1241651041_

rhaschke, 2022-09-09:
You are straightening the arm to its full length. This pushes the arm into a singularity, which automatically causes joint-space jumps. That's normal and cannot be avoided. Simply avoid reaching beyond the workspace!

thomasw2, 2022-09-09:
Thanks! That is likely the source of the problem. Will check whether that is the case for the UR3e.