moveit/moveit #3214: Noetic release

rhaschke, 2022-09-12:
Dear @simonschmeisser and @v4hn,
I'm planning to prepare a new MoveIt release for Noetic within this week - the last one was in March. Any thoughts on what should definitely go in there?

simonschmeisser, 2022-09-12:
That would be great, if only for incrementing the version number and allowing to ifdef more easily!

I don't know of any open show stoppers, issue or PRs, please go ahead as your time and motivation permits.

v4hn, 2022-09-12:
Definitely a good idea!
I agree with Simon. They're might be relevant patches again relatively soon (GSoC, maybe things related to the mbs,.. ) but right now I'm not aware of anything urgent.
Go ahead when you have time or tell us if you want a quick review for something or yours.

rhaschke, 2022-09-13:
> If only for incrementing the version number and allowing to ifdef more easily

Did you have a look at #3211? That would provide an automated way to increase the version number of #defines.

rhaschke, 2022-09-13:
Noetic: https://github.com/ros/rosdistro/pull/34461
Melodic: https://github.com/ros/rosdistro/pull/34458

closed 2022-09-13 by rhaschke