moveit/moveit #3222: Python interface not planning to attached collision object subframe

vinceHuyghe, 2022-09-22:
### Description

When trying to plan with an attached collision object’s subframe set as the end effector link the following error is thrown: 
ERROR: Link 'end_effector/tcp' not found in model 'fanuc'

### Your environment
* ROS Distro: Noetic 
* OS Version: e.g. Ubuntu 20.04
* Binary build
* latest moveit version

### Steps to reproduce
I have created a minimal example in following [repo](https://github.com/vinceHuyghe/example) this has been tested with [moveit_resources](https://github.com/ros-planning/moveit_resources) fanuc_moveit_config demo.launch and reproduces the error.

### Actual behaviour
ERROR: Link 'end_effector/tcp' not found in model 'fanuc'

rhaschke, 2022-09-30:
Your example works for me w/o any warnings or errors. Do you have the `ResolveConstraintFrames` planning adapter activated?

rhaschke, 2022-09-30:
This behavior resulted from a check to avoid planning if all goal constraints are already satisfied. Unfortunately, this check doesn't have the sub frames resolved yet and thus throws this error. I filed a simple fix: #3228.
Meanwhile, you can simply avoid the problem by planning and executing separately - `go()` performs both steps internally and your `plan()` call was actually redundant:

```python
        th.move_group.set_pose_target(pose)
        success, plan = th.move_group.plan()[:2]
        if success:
            th.move_group.execute(plan)
```

vinceHuyghe, 2022-10-03:
Thank you. I will close the issue.

PR #3228 (2022-10-05): Drop shortcut avoiding planning when all constraints are already met
When calling the PlanAndExcute action, there was a shortcut avoiding planning (and execution) at all when all goal constraints were already met. However, this check also succeeded when there was an error, e.g. resolving link names. As proper link resolution might require some planning request adapters to be executed, e.g. ResolveConstraintFrames, we should not skip planning.

Note, calling the PlanOnly action, didn't have this shortcut as well, resulting in inconsistent behavior.

I just decided to remove the shortcut check c [... cut]
closed 2022-10-03 by vinceHuyghe