moveit/moveit #3225: how can we use  moveit to  remove robot arm from  the pointcloud  generated by 3d camera mounted outside the arm when build octomap

dbdxnuliba, 2022-09-28:
how can we use  moveit to  remove robot arm from  the pointcloud  generated by 3d camera mounted outside the arm when build octomap

rhaschke, 2022-11-07:
> but what if we has not the target model

Which target model? The robot model? If you don't have that, how to you specify your URDF? 
Obviously, you need to have the robot model, to hide it from the octomap.

dbdxnuliba, 2022-11-12:
> > but what if we has not the target model
> 
> Which target model? The robot model? If you don't have that, how to you specify your URDF? Obviously, you need to have the robot model, to hide it from the octomap.

The target model is object model ,such as object to be grasped ,but we but what if we has not the 3D-model of the object to be grasped, how can we remove the object from the octomap ,

nyxrobotics, 2023-08-09:
If the point cloud of the robot body is visible, is there an option to ignore them in octomap? Also, I think we need a padding option because our robot arm is not calibrated well enough.
[image]

rhaschke, 2023-08-09:
There is an official MoveIt tutorial explaining robot-mesh filtering:
https://ros-planning.github.io/moveit_tutorials/doc/mesh_filter/mesh_filter_tutorial.html

nyxrobotics, 2024-04-23:
> There is an official MoveIt tutorial explaining robot-mesh filtering: https://ros-planning.github.io/moveit_tutorials/doc/mesh_filter/mesh_filter_tutorial.html

Thank you so much.
I was able to remove the robot point cloud by following your instructions, but I have one question.

In `depth_self_filter_nodelet.cpp`, there is a function to remove the robot's mesh, which I was able to use.

On the other hand, the description of the parameter `filtered_cloud_topic` in `DepthImageOctomapUpdater` seems to implement the same function, but I could not find any keywords such as link or robot_model in `depth_image_octomap_updater.cpp`.

Can you tell me where the self filter is implemented in the `DepthImageOctomapUpdater`?