moveit/moveit #3242: Transform error of sensor data: Lookup would require extrapolation

dbdxnuliba, 2022-10-12:
ubuntu20.04 noetic
I use head_camera and torso camera D435 publish point_cloud and one arm on the rail
after I config moveit ,and Launch it
the terminal in the moveit show the Error like that:

[image]
[ERROR] [1665540460.474683451]: Transform error of sensor data: Lookup would require extrapolation 0.414198207s into the past.  Requested time 1665540450.045501709 but the earliest data is at time 1665540450.459699869, when looking up transform from frame [camera2_color_optical_frame] to frame [world]; quitting callback
[ERROR] [1665540462.095226504]: Transform error: Lookup would require extrapolation 0.439864105s into the past.  Requested time 1665540451.633910894 but the earliest data is at time 1665540452.073775053, when looking up transform from frame [left_arm_link5] to frame [camera1_color_optical_frame]
[ERROR] [1665540462.100130489]: Transform cache was not updated. Self-filtering may fail. If transforms were not available yet, consider setting robot_description_planning/shape_transform_cache_lookup_wait_time to wait longer for transforms
[ERROR] [1665540469.518701594]: Transform error of sensor data: Lookup would require extrapolation 0.112081074s into the past.  Requested time 1665540459.419473410 but the earliest data is at time 1665540459.531554461, when looking up transform from frame [camera2_color_optical_frame] to frame [world]; quitting callback
[ERROR] [1665540469.572366942]: Transform error: Lookup would require extrapolation 0.092123858s into the past.  Requested time 1665540459.473356485 but the earliest data is at time 1665540459.565480232, when looking up transform from frame [left_arm_link5] to frame [camera1_color_optical_frame]
[ERROR] [1665540469.581037343]: Transform cache was not updated. Self-filtering may fail. If transforms were not available yet, consider setting robot_description_planning/shape_transform_cache_lookup_wait_time to wait longer for transforms

tf_tree
[image]

rhaschke, 2022-10-12:
This error stems from this source:
https://github.com/ros-planning/moveit/blob/ae6e7bf79c8c0efcd463ab98c094c07d04a4f747/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L207-L208

and originates from an exception in TF lookup thrown if TF data needs to be extrapolated beyond the timestamps available in the TF cache. Please check that your point cloud publisher (defining the lookup timestamp) actually updates the timestamps.
Maybe you have that error only initially when there is no matching frame info available yet for the `torso_lift_link` (which is the only non-static link in the TF tree between `world` and `camera2_color_optical_frame`)?

Kubasinska, 2022-10-25:
I have the same problem with the same cameras. Did you find a solution?

dbdxnuliba, 2022-11-07:
@Kubasinska @rhaschke ,I has not solved the problem up to now

rhaschke, 2022-11-07:
> The same lag is present in the robot:

@Kubasinska: a small time lag is normal. Here the lag is -1s + 999543966ns = 456µs

@ both of you: Do you see the TF warnings continuously or only once at the beginning?
Please check the time delay between your camera data stream and ROS time.

Kubasinska, 2022-11-07:
I found that (in my case) the problem is present **exclusively when the actual robot enters the camera frame**. For now I have circumvented the problem. In fact, by using the DepthImageOctomapUpdater instead of the PointCloudOctomapUpdater (in sensor_3d.yaml) I have no problem.