moveit/moveit #3245: Failed to Fetch current robot states

Dafke08, 2022-10-18:
Hello,

I am currently working on a Niryo Ned Robot. 

I want to make a simple Pick and Place machine by running a Python script.
The task is need to do is move to left. Go down 5 cm. And move back.

However if I want to let the robot move downwards 5 cm by using the .get_current_Pose() and modify it. I get the follow error:

Failed to fetch current robot state

I Am using the following code:
#! /usr/bin/env python

import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import actionlib
import geometry_msgs

def simple_pick_place():
 	moveit_commander.roscpp_initialize(sys.argv)
 	rospy.init_node('simple_pick_place',
                  anonymous=True)

 	robot1_group = moveit_commander.MoveGroupCommander("arm")

	robot1_client = actionlib.SimpleActionClient ('execute_trajectory',moveit_msgs.msg.ExecuteTrajectoryAction)
	robot1_client.wait_for_server()
	rospy.loginfo ('Execute Trajectory server is available for robot1')

	robot1_group.set_named_target ("home")

	robot1_plan_home = robot1_group.plan()
	robot1_goal = moveit_msgs.msg.ExecuteTrajectoryGoal()
	robot1_goal.trajectory = robot1_plan_home

	robot1_client.send_goal(robot1_goal)
	robot1_client.wait_for_result()

	robot1_group.set_named_target ("left")

	robot1_plan_home = robot1_group.plan()
	robot1_goal = moveit_msgs.msg.ExecuteTrajectoryGoal()
	robot1_goal.trajectory = robot1_plan_home

	robot1_client.send_goal(robot1_goal)
	robot1_client.wait_for_result()
	
	## Cartesian Paths
	waypoints = []
	# start with the current pose
	current_pose = robot1_group.get_current_pose()
	rospy.sleep(0.5)
	current_pose = robot1_group.get_current_pose()
	
	## create linear offsets to the current pose
	new_eef_pose = geometry_msgs.msg.Pose()

	
	new_eef_pose.position.z = current_pose.pose.position.z -5

	new_eef_pose.orientation = copy.deepcopy(current_pose.pose.orientation)

	waypoints.append(new_eef_pose)
	print(current_pose.pose.position)
	print(new_eef_pose.position)
	fraction = 0.0
	for count_cartesian_path in range(0,3):
	  if fraction < 1.0:
	     (plan_cartesian, fraction) = robot1_group.compute_cartesian_path(
		                           waypoints,   # waypoints to follow
		                           0.01,        # eef_step
		                           0.0)         # jump_threshold
	  else:
	      break

	rospy.loginfo('x')
	robot1_goal = moveit_msgs.msg.ExecuteTrajectoryGoal()
	robot1_goal.trajectory = plan_cartesian
	robot1_client.send_goal(robot1_goal)
	robot1_client.wait_for_result()

	robot1_group.set_named_target ("home")

	robot1_plan_home = robot1_group.plan()
	robot1_goal = moveit_msgs.msg.ExecuteTrajectoryGoal()
	robot1_goal.trajectory = robot1_plan_home

	robot1_client.send_goal(robot1_goal)
	robot1_client.wait_for_result()

	robot1_group.set_named_target ("resting")

	robot1_plan_home = robot1_group.plan()
	robot1_goal = moveit_msgs.msg.ExecuteTrajectoryGoal()
	robot1_goal.trajectory = robot1_plan_home

	robot1_client.send_goal(robot1_goal)
	robot1_client.wait_for_result()

if __name__=='__main__':
  try:
    simple_pick_place()
  except rospy.ROSInterruptException:
    pass

I have searched for the issue and see only cpp solutions, but I am using Python. So how can i work around this problem?

Thanks in advance

Sanraj-Lachhiramka, 2024-09-10:
Hey, did u find solution for this. I am facing the same issue

rhaschke, 2024-09-10:
The error message `Failed to fetch current robot state` is thrown if the current state cannot be retrieved within a reasonable amount of time. You should also get an info message on the console explaining possible causes:
```
Didn't receive robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
```
Is your robot publishing its full state frequently?

Sanraj-Lachhiramka, 2024-09-11:
I am getting this warning

[ WARN] [1726030907.336028979, 85.167000000]: Failed to fetch current robot
state.
[ WARN] [1726030907.336632176, 85.167000000]: Unable to transform object
from frame 'TCP' to planning frame 'world' (Could not find a connection
between 'world' and 'TCP' because they are not part of the same tree.Tf has
two or more unconnected trees.)

rhaschke, 2024-09-11:
You didn't reply to my question: Is your robot publishing its full state frequently? At which rate? Are all joints included?