moveit/moveit #3246: xception while loading planner 'trajopt_interface/TrajOptPlanner': According to the loaded plugin descriptions the class trajopt_interface/TrajOptPlanner with base class type planning_interface::PlannerManager does not exist

dbdxnuliba, 2022-10-19:
### Description

Overview of your issue here.

### Your environment
* ROS Distro:Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? source
* If source, which branch ?moveit-1.1.9

### Steps to reproduce
roslaunch panda_moveit_config demo.launch  pipeline:=trajopt
[image]
[image]

[ INFO] [1666163881.715930689]: Loading planning pipeline 'trajopt'
[ERROR] [1666163881.717387351]: Exception while loading planner 'trajopt_interface/TrajOptPlanner': According to the loaded plugin descriptions the class trajopt_interface/TrajOptPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  chomp_interface/CHOMPPlanner lerp_interface/LERPPlanner ompl_interface/OMPLPlanner pilz_industrial_motion_planner::CommandPlanner
Available plugins: chomp_interface/CHOMPPlanner, lerp_interface/LERPPlanner, ompl_interface/OMPLPlanner, pilz_industrial_motion_planner::CommandPlanner

[ERROR] [1666163881.719558047]: Failed to initialize planning pipeline 'trajopt'.
[ INFO] [1666163881.725016287]: Loading robot model 'panda'...
[ INFO] [1666163881.746353253]: Set joints of group 'panda_arm' to pose 'ready'.
[ INFO] [1666163881.746406194]: Set joints of group 'panda_hand' to pose 'open'.
[ INFO] [1666163881.746663678]: Fake controller 'fake_panda_arm_controller' with joints [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ INFO] [1666163881.746874786]: Fake controller 'fake_panda_hand_controller' with joints [ panda_finger_joint1 ]
[ INFO] [1666163881.747040443]: Returned 2 controllers in list
[ INFO] [1666163881.751139149]: Trajectory execution is managing controllers
[ INFO] [1666163881.751162597]: MoveGroup debug mode is ON
[ERROR] [1666163881.751444293]: Failed to find default PlanningPipeline 'trajopt' - please check MoveGroup's planning pipeline configuration.
[ INFO] [1666163881.826893727]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[move_group-5] process has died [pid 136590, exit code -11, cmd /home/taimi/ref/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/taimi/.ros/log/29e48136-4f7e-11ed-91b1-471f2b8550a2/move_group-5.log].
log file: /home/taimi/.ros/log/29e48136-4f7e-11ed-91b1-471f2b8550a2/move_group-5*.log
[ INFO] [1666163885.015790240]: Loading robot model 'panda'...
[ INFO] [1666163885.077284116]: Starting planning scene monitor
[ INFO] [1666163885.079171261]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1666163885.298943898]: waitForService: Service [/get_planning_scene] could not connect to host [taimi-NUC8i7BEH:53563], waiting...
[ INFO] [1666163890.299286159]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1666163890.299390680]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1666163920.303855367]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
[ INFO] [1666163920.304212837]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1666163950.308593055]: Unable to connect to move_group action server 'move_group' within allotted time (30s)

yzy244001743, 2022-12-23:
I have the same problem. Have you solved it ？

simonschmeisser, 2022-12-23:
Building the trajopt plugin is currently disabled as it is far from functional in it's current state. This is done by this file: https://github.com/ros-planning/moveit/blob/master/moveit_planners/trajopt/CATKIN_IGNORE
If you build from source you can delete this file and experiment with the plugin and contribute improvements. Please see also #3235 for a slightly improved version that is still very much WIP though.

closed 2022-12-23 by simonschmeisser

zengyuhan320, 2023-01-11:
I have the same problem. Have you solved it ？

ChuGuoqing, 2023-04-23:
I have the same problem. Have you solved it ？

for1314ai, 2024-11-15:
粑粑一样