moveit/moveit #3247: error ,when roslaunch moveit_tutorials trajopt_example_launch.launch

dbdxnuliba, 2022-10-19:
### Description

Overview of your issue here.

### Your environment
* ROS Distro:|Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? source

* If source, which branch? moveit-1.1.9
1)roslaunch panda_moveit_config demo.launch pipeline:=trajopt
2)roslaunch moveit_tutorials trajopt_example_launch.launch
in the second terminal it shows error info:

ERROR Initial trajectory is not the right size matrix
Expected 20 rows (time steps) x 7 columns
Got 7 rows and 1 columns
at /home/taimi/ref/ws_moveit/src/moveit-1.1.9/moveit_planners/trajopt/src/problem_description.cpp:212
[ERROR] [1666168302.758164399]: Exception caught: 'Initial trajectory is not the right size matrix
Expected 20 rows (time steps) x 7 columns
Got 7 rows and 1 columns'

[image]

elrond91, 2022-11-25:
have the same error, somebody have fix it ?

zengyuhan320, 2023-01-13:
hi~,I have the same error, anyone have fix it ?

sleichu2, 2023-01-15:
same err,somebody has fixed it?

simonschmeisser, 2023-01-15:
I added some fixes in #3230 which should be part of the latest release and there is a WIP PR #3235 you could give a try. Please stop the spam and use the reactions (+1 or similar) if you don't have anything else to say besides me too

Why are you using 1.1.9 for example if you build from source?

sleichu2, 2023-02-06:
These fixes in #3230 and #3235 had no work.Could you give me some advise about this error.Is it a library fault or is there a configuration error in the example?

simonschmeisser, 2023-02-06:
You could test 
```
init_info:
 type: 2 #JOINT_INTERPOLATED
```
maybe initial trajectory mode is broken. How do you set the initial trajectory?

interpolation is implemented here: https://github.com/ros-planning/moveit/blob/b32af6a22a134f8bb150d659ca14b93ab4617adf/moveit_planners/trajopt/src/problem_description.cpp#L575

initial trajectory would need to be implemented here:
https://github.com/ros-planning/moveit/blob/b32af6a22a134f8bb150d659ca14b93ab4617adf/moveit_planners/trajopt/src/trajopt_interface.cpp#L142

currently it looks like the intention was to just copy the start position n times

sleichu2, 2023-02-07:
Worked!! Thanks a lot.

sleichu2, 2023-02-07:
Sorry to disturb you again, but the planner is currently unable to achieve obstacle avoidance, as printed below:

<pre>[ INFO] [1675752591.723613206]:  ==================================== Response
[ INFO] [1675752591.723627302]:  ==================================== Debug Response
[ INFO] [1675752591.723635087]: 
[ INFO] [1675752591.723644455]: 12
[ INFO] [1675752591.723652260]: 12
[ INFO] [1675752591.723660736]: 6
[ INFO] [1675752591.723667719]: 20
<font color="#CC0000">[ERROR] [1675752591.724331254]: Computed path is not valid. Invalid states at index locations: [ 8 9 10 11 12 13 14 15 16 17 18 ] out of 20. Explanations follow in command line. Contacts are published on /move_group/display_contacts</font>
[ INFO] [1675752591.724357243]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_left_finger_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724366551]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724451281]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_left_finger_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724460469]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724541923]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_left_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724551531]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724615261]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_left_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724624018]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724684172]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_inner_knuckle_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724692878]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724756869]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724765436]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724825910]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724834647]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724892827]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724901443]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.724968359]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.724977046]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.725057859]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;robotiq_85_right_finger_tip_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.725066946]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1675752591.725151787]: Found a contact between &apos;box1&apos; (type &apos;Robot attached&apos;) and &apos;forearm_link&apos; (type &apos;Robot link&apos;), which constitutes a collision. Contact information is not stored.
[ INFO] [1675752591.725160794]: Collision checking is considered complete (collision was found and 0 contacts are stored)
<font color="#CC0000">[ERROR] [1675752591.725201430]: Completed listing of explanations for invalid states.</font>
<font color="#C4A000">[ WARN] [1675752591.743441209]: Fail: ABORTED: Motion plan was found but it seems to be invalid (possibly due to postprocessing). Not executing.</font></pre>