moveit/moveit #3249: Moveit Commander go

mathgameapi, 2022-10-20:
I pass the goal joints to go function.
I found the log "Collision checking is considered complete (collision was found and 0 contacts are stored)"

But the robot arm still execute. Finial the arm collision with object.

How can make to not execute when collision was found?
Thanks.

Akumar201, 2022-10-20:
Hi can you please share snippet of your code for better understanding.

mathgameapi, 2022-10-21:
> Hi can you please share snippet of your code for better understanding.

joint_state = JointState()
joint_state.name = ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint',
                                'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
joint_state.position = joint_goal
success, traj, planning_time, error = self._move_group.plan(
                joint_state)
if success:
   success = self._move_group.execute(traj, wait=wait)
   
 Sometime i got,
"Collision checking is considered complete (collision was found and 0 contacts are stored)"
"Start state appears to be in collision with respect to group arm "
"Found a valid state near the start state at distance 2.540824 after 1 attempts"

At that time, the arm will have unexpected movement. 

Thanks