moveit/moveit #3256: "Unable to identify any set of controllers that can actuate the specified joints: ..." Gazebo Simulation

TKroos18, 2022-10-29:
### Description
I'm learning how to use moveit. After following the MoveIt setup assistant tutorial steps..., when I tried to execute `roslaunch yourobot_moveit_config demo_gazebo.launch,` I got error messages. I opened my `ros_controllers.yaml` and found that there was very little content in the file.
My ros_controllers.yaml:
`meca_arm_controller:
  type: position_controllers/JointTrajectoryController
  joints:
    - meca_axis_1_joint
    - meca_axis_2_joint
    - meca_axis_3_joint
    - meca_axis_4_joint
    - meca_axis_5_joint
    - meca_axis_6_joint
  gains:
    meca_axis_1_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
    meca_axis_2_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
    meca_axis_3_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
    meca_axis_4_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
    meca_axis_5_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1
    meca_axis_6_joint:
      p: 100
      d: 1
      i: 1
      i_clamp: 1`

And my ros_controller.launch:
`<?xml version="1.0"?>
<launch>

  
  <rosparam file="$(find mecademic_moveit_config)/config/ros_controllers.yaml" command="load"/>

  
  <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="meca_arm_controller joint_state_controller"/>

</launch>`

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Binary Build, 1.1.10-1focal.20221003.150826 amd64

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output
[controller_spawner-6] process has finished cleanly
log file: /home/cqsubuntu/.ros/log/64765f3c-5766-11ed-a11a-873155141b87/controller_spawner-6*.log
[ INFO] [1667033314.503774496, 29.101000000]: Execution request received
**[ WARN] [1667033314.504965238, 29.102000000]: Failed to read controllers from /controller_manager/list_controllers
[ERROR] [1667033314.505176744, 29.102000000]: Unable to identify any set of controllers that can actuate the specified joints: [ meca_axis_1_joint meca_axis_2_joint meca_axis_3_joint meca_axis_4_joint meca_axis_5_joint meca_axis_6_joint ]
[ERROR] [1667033314.505306038, 29.102000000]: Known controllers and their joints:**

[ INFO] [1667033314.524839334, 29.122000000]: ABORTED: Solution found but controller failed during execution

rhaschke, 2022-10-29:
Is your `ros_controllers.yaml` loaded by any launch file? Compare other moveit_config packages!

TKroos18, 2022-10-29:
> Is your `ros_controllers.yaml` loaded by any launch file? Compare other moveit_config packages!
I think it is loaded by ros_controllers.launch. When I execute demo_gazebo.launch, ros_controllers.launch should be executed.
I had tried to compare other packages but they look too complicated to find differences...
[moveitgazebo_test_pro.zip](https://github.com/ros-planning/moveit/files/9893404/moveitgazebo_test_pro.zip)

TKroos18, 2022-10-29:
> Is your `ros_controllers.yaml` loaded by any launch file? Compare other moveit_config packages!

I think it was loaded because "demo_gazebo.launch -> gazebo.launch -> ros_controllers.launch -> ros_controllers.yaml".
I cloned https://github.com/ros-planning/panda_moveit_config and tried to compare it with mine. I found some differences. In panda_moveit_config, "moveit_controller_manager" is "simple" . In my package, it is "ros_control".[Uploading ros_empty_pro.zip…]()
I uploaded my project. Maybe it can describe my problem more clearly.

rhaschke, 2022-10-30:
You have several issues in your config:
1. Gazebo's controller_manager is loaded in a namespace, which is why the controller spawner could not find it:
   `[WARN] rosout: Controller Spawner couldn't find the expected controller_manager ROS interface.`
2. You load the `joint_state_controller` twice
3. You don't yet define control gains:
   ```
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_1_joint
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_2_joint
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_3_joint
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_4_joint
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_5_joint
   [ERROR] ros.control_toolbox: No p gain specified for pid.  Namespace: /meca_500_r3/gazebo_ros_control/pid_gains/meca_axis_6_joint
   ```
All this was indicated via corresponding warning and error messages in the console before starting any MoveIt interaction!
You should consider those warnings and fix them. Please find patches for the first two issues here: [patches.zip](https://github.com/ros-planning/moveit/files/9896127/patches.zip)
If that was helpful, please consider [donating](https://github.com/sponsors/rhaschke).

TKroos18, 2022-12-02:
Thank you very much! I think I have known what happened...